chipsalliance/verible #178: syntax error (unexpected EOF)

weicaiyang, 2020-02-04:
seems like the verilog_lint can't handle included file very well. We have a file ([link](https://github.com/google/riscv-dv/blob/769e1cf6d646bbc40886384a36688a3cbbcf7139/src/isa/custom/riscv_custom_instr_enum.sv)), which doesn't contain completed task/function. And we got below error for it.

> src/isa/custom/riscv_custom_instr_enum.sv:3:1: syntax error (unexpected EOF) (https://github.com/google/verible).

fangism, 2020-02-04:
That file looks like it just has two lines, without very much context.  It doesn't even resemble a function/task body.  The comment suggests it is an enum body.  You can select parsing modes to assume different contexts, but we haven't added one for enum body yet (is that what you want?).  Here is the list of available contexts:
https://github.com/google/verible/blob/ca3fce403ab5bf2cf56795b747da718c2be2b3ca/verilog/analysis/verilog_excerpt_parse.cc#L106

e.g. you can use these like:

```
// verilog_syntax: parse-as-module-body
module_contents go_here;
```

weicaiyang, 2020-02-04:
> That file looks like it just has two lines, without very much context. It doesn't even resemble a function/task body. The comment suggests it is an enum body. You can select parsing modes to assume [...]

I feel it would be nice if we can parse the whole thing, like this one with the file including it.

fangism, 2020-02-04:
re: "parse the whole thing"
At this time, the Verible parser is only solely focused on single-file operations, and doesn't even have a proper preprocessor yet (which is what you would need to expand `` `includes `` and expand macros).  I'm afraid you might need to ignore analyzing file snippets that don't fall into one of the existing parsing modes.  

That being said, if you do have a standalone preprocessor from another toolchain, you could feed the result of that into `verilog_lint`, however, file-line tracking through expansion isn't supported yet.

weicaiyang, 2020-02-04:
> re: "parse the whole thing"
> At this time, the Verible parser is only solely focused on single-file operations, and doesn't even have a proper preprocessor yet (which is what you would need to expand [...]

I see. Thanks for the explanation. 
Yes, I have added it into our ignored list to avoid the error.

fangism, 2020-02-05:
Closing this as won't fix for now, not at least until we have a proper preprocessor.

closed 2020-02-05 by fangism