chipsalliance/verible #184: Abstract Syntax Tree (AST)

fangism, 2020-02-05:
Verible's parser today yields a concrete syntax tree (CST), which views all syntactic tokens (including punctuation and sugar) into as a tree (comments are preserved in a separate array).  The CST is a language-agnostic structure with language-specific markers (in Verible, node enumerations).  This was good enough to quickly build a style linter and code formatter.  The CST structure itself is fragile and prone to change and semi-arbitrary shaping.  Real compiler toolchains operate on ASTs.  

Wish:
* An *implementation-neutral* AST class library that *any* parser tool could produce, and any front-end compiler toolchain could consume.  It is not important that Verible owns the said AST.  Ideally, this would come from a community of tool developers and meet everyone's requirements.  
* AST validator, that can further diagnose invalid code in ways that was not possible (or too difficult) with a plain CST, or raw parser.

Ideas:
* The https://github.com/google/verible/blob/master/verilog/parser/verilog.y yacc grammar and implementation includes preprocessing constructs.  If we stripped those out, we'd have a parser suitable for handling preprocessed code.  It could share the same CST output (even node enumerations), but just be devoid of preprocessing constructs, and suitable for transformation to AST.

See also #183 regarding a preprocessor.

Ping this issue if interested.

amaleewilson, 2021-12-15:
Hi, I'm looking for a System Verilog parser that can be modified to emit one of the hardware dialects in CIRCT (https://github.com/llvm/circt). I'd really like to extend Verible's parser, but the lack of an AST makes that difficult. Has there been any work on this?

fangism, 2021-12-16:
Hi!  These page describes what we have for a syntax tree at the moment:

* https://github.com/chipsalliance/verible/tree/master/verilog/tools/syntax
* https://github.com/chipsalliance/verible/tree/master/verilog/tools/syntax/export_json_examples

The "tree" is not abstract; it is a structural representation that includes syntactic sugar and punctuation tokens.  The structure is ad hoc, not guarantee to be stable at this time, and there are no syntax tree node classes.  That said, it can still be very useful for many applications.  

Are the operations you have in mind mechanical syntax translations, or do they require semantics?