chipsalliance/verible #185: source code indexing, symbol/fact extraction, cross-reference linking

fangism, 2020-02-05:
Gist:
* extract symbol definitions and declarations and references
* establish use/definition links
* symbols would be properly scoped

Purpose:
* code navigation (IDEs)
* auto-completion
* all-of-the-above, but incremental, and operable on local workspace changes

Representation? any structure that is easily exportable/convertible to interfaces expected by other tools such as IDEs.
For example https://github.com/kythe/kythe is one platform for serving source code information and metadata.

fangism, 2020-08-05:
@MinaToma work is ongoing at https://github.com/MinaToma/verible

fangism, 2020-08-06:
@ivan444 FYI

fangism, 2020-08-06:
b/163065644

ivan444, 2020-10-29:
This is somewhat done. The indexing is working. The remaining tasks are covered with the finer grained bugs. This issue is not useful anymore.

closed 2020-10-29 by ivan444