chipsalliance/verible #187: undeclared inclusion(s) in rule '@com_google_absl//absl/numeric:int128'

mithro, 2020-02-06:
07d75ca426710889de9c22ffd23e4ecd62001172
```
ERROR: /home/tansell/.cache/bazel/_bazel_tansell/df58af9066c94d9bc43127b2a3871557/external/com_google_absl/absl/numeric/BUILD.bazel:26:1: undeclared inclusion(s) in rule '@com_google_absl//absl/numeric:int128':
this rule is missing dependency declarations for the following files included by 'external/com_google_absl/absl/numeric/int128.cc':
  '/usr/lib/gcc/x86_64-linux-gnu/9/include/stddef.h'
  '/usr/lib/gcc/x86_64-linux-gnu/9/include/stdint.h'
  '/usr/lib/gcc/x86_64-linux-gnu/9/include/stdarg.h'
  '/usr/lib/gcc/x86_64-linux-gnu/9/include-fixed/limits.h'
  '/usr/lib/gcc/x86_64-linux-gnu/9/include-fixed/syslimits.h'
Target //verilog/tools/lint:verilog_lint failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 7.476s, Critical Path: 5.06s
INFO: 0 processes.
FAILED: Build did NOT complete successfully
make: *** [Makefile:15: bazel-bin/verilog/tools/lint/verilog_lint] Error 1
```

hzeller, 2020-02-06:
Remove the bazel cache
```
mv ~/.cache/bazel ~/.cache/bazel-old
```

... if it has compiled something before and the compiler version changed, it is confused.

fangism, 2020-05-19:
@mithro obsolete?

mithro, 2020-05-19:
I think so.

closed 2020-05-19 by mithro