chipsalliance/verible #192: Integration of verilog_lint into various editors (how-to documentation)

fangism, 2020-02-11:
If you know how to plug a custom analyzer tool into your favorite editor to give instant feedback as you type, we need your help.  Scripts and/or documentation would all be appreciated.

Doix, 2020-02-11:
Just posting this in case it helps someone else out there. If someone is a vim & ale (https://github.com/dense-analysis/ale) user, then this should be enough to get it kinda working: 

```
if !exists('g:ale_verilog_verible_options')
    let g:ale_verilog_verible_options = ''
endif

function! ale_linters#verilog#verible#GetCommand(buffer) abort
    let l:filename = tempname() . '_verible_linted.sv'

    " Create a special filename, so we can detect it in the handler.
    call ale#engine#ManageFile(a:buffer, l:filename)
    let l:lines = getbufline(a:buffer, 1, '$')
    call ale#util#Writefile(a:buffer, l:lines, l:filename)

    return 'verilog_lint '
    \   . ale#Var(a:buffer, 'verilog_verible_options') .' '
    \   . ale#Escape(l:filename)
endfunction

function! ale_linters#verilog#verible#Handle(buffer, lines) abort
    " Look for lines like the following.
    "
    " filename.sv:7:13: Explicitly define static or automatic lifetime for non-class functions [Style: function-task-explicit-lifetime] [explicit-function-lifetime]
    let l:pattern = '\([A-z0-9]\+\.sv\):\(\d\+\):\d\+:\(.*\s\+\)\([Style:.*\)'
    let l:output = []
    for l:match in ale#util#GetMatches(a:lines, l:pattern)
        let l:file = l:match[1]
        let l:line = l:match[2]
        let l:text = l:match[3]
        let l:type = l:match[4]

        if l:file =~# '_verible_linted.sv'
            call add(l:output, {
            \   'lnum': l:line,
            \   'text': l:text,
            \   'type': l:type,
            \})
        endif
    endfor

    return l:output
endfunction

call ale#linter#Define('verilog', {
\   'name': 'verible',
\   'output_stream': 'stdout',
\   'executable': 'verilog_lint',
\   'command_callback': 'ale_linters#verilog#verible#GetCommand',
\   'callback': 'ale_linters#verilog#verible#Handle',
\   'read_buffer': 0,
\})
```
Just stick that in a file at:<vim_plugin_path>/ale/ale_linters/verilog/verible.vim. I know very little about vimscript, I just took one of the other linters and s/LINTER/verible/ and fixed up the regex it uses. Hopefully someone with better vimscript knowledge could create a pull request for the ale project and get it merged in mainline.

I suspect setting verilog_format as the fixer for verilog would work as well:
```
let g:ale_fixers = {'verilog': ['verilog_format']}
```

fangism, 2020-02-11:
Thanks @Doix any information helps!

vhda, 2020-05-01:
@Doix why not create the PR and wait for feedback from ALE's maintainers?

fangism, 2020-05-01:
For those interested, we're considering enabling some fixer-like features into the linter in #284 .  These would be intended address non-whitespace changes, whereas the formatter would stick to whitespace-only changes.

zegervdv, 2020-05-28:
I hadn't seen this issue before, but today I just launched a PR to add verible to ale (dense-analysis/ale#3192)

I started from the verilator, seems like @doix did the same so it looks very similar.

rdburns, 2020-12-29:
I have it working with Emacs via FlyCheck, and here is the config:
``` lisp
;; Verible linter doesn't have --stdin flag, so need to use temporary file
;; I couldn't get the directory traversal working to find the config file so I spelled it out exactly

(flycheck-define-checker verible-verilog-lint
  "A SystemVerilog syntax and style checker using the Verible linter tool from Google.

See URL `https://github.com/google/verible'."
  :command ("verible-verilog-lint" source "--rules_config=/home/youruser/.rules.verible_lint")
  :standard-input f
  :modes (verilog-mode)
  :error-patterns
  (
    (error line-start (file-name) ":" line ":" column ": syntax error, " (message) line-end)
    (info line-start (file-name) ":" line ":" column ": " (message) line-end)
    ;; You could change the severity of lint messages to warnings if you wish
  )

)

(add-to-list 'flycheck-checkers 'verible-verilog-lint 'append)
(setq flycheck-checker-error-threshold 10000)
```

fangism, 2020-12-30:
@rdburns Thanks for the emacs tip!
If you pass `-` as a file name to `verible-verilog-lint` that will read from stdin.  
Does that help you avoid the temporary file?

reportaman, 2021-05-23:
@Doix Doing what you suggest gives me error in vim & nevim:

```
Error detected while processing function ale#events#LintOnEnter[5]..ale#Queue[33]..<SNR>129_Lint[3]..ale#linter#Get:
line   15:
E745: Using a List as a Number
E15: Invalid expression: [l:linter.name] + l:linter.aliases
```

I tried both settings:
```
let g:ale_linters = {
\   'verilog': ['verilog_lint']
\}
```
and
```
let g:ale_linters = {
\   'verilog': ['verible']
\}
```

Also, where do you set the name of the binary ``verible_verilog_lint`` in your code??

Doix, 2021-05-23:
So by default all linters should be enabled for your filetype and I haven't tried explicitly only enabling verible. The name needs to match the name of your .vim file. So if you called the file verible.vim it should be 

```
let g:ale_linters = {
\   'verilog': ['verible']
\}
```
But I've never tried setting that. What I do have, is this:
```
let g:ale_linter_aliases = {'verilog_systemverilog': 'verilog'}
```
Which would help if you are using verilog_systemverilog from: https://github.com/vhda/verilog_systemverilog.vim 

But I don't think that's your problem. I think the problem is that ale has changed since I made my comment and this line is out of date:
```
\   'command_callback': 'ale_linters#verilog#verible#GetCommand',
``` 
it instead should be:
```
\   'command': function('ale_linters#verilog#verilator#GetCommand'),
```

>Also, where do you set the name of the binary verible_verilog_lint in your code??
I guess the name of the binary changed at some point, it's at the bottom of the code snippet I linked:
```
\   'executable': 'verilog_lint',
```
Change that to the proper executable name. 

I haven't written verilog in a while, so I haven't been keeping up to date with all these changes. Sorry for the hassle, I hope that helps.

reportaman, 2021-05-23:
@Doix It has another error now. First my exact settings:

In ``.vimrc``:
```vim
let g:ale_linter_aliases = {'verilog_systemverilog': 'verilog'}
let g:ale_linters = {
\   'python': ['pylint'],
\   'verilog': ['verible']
\}
```

In ``/Users/reportaman/.vim/plugged/ale/ale_linters/verilog/verible.vim``:
```vim
if !exists('g:ale_verilog_verible_options')
    let g:ale_verilog_verible_options = ''
endif

function! ale_linters#verilog#verible#GetCommand(buffer) abort
    let l:filename = tempname() . '_verible_linted.sv'

    " Create a special filename, so we can detect it in the handler.
    call ale#engine#ManageFile(a:buffer, l:filename)
    let l:lines = getbufline(a:buffer, 1, '$')
    call ale#util#Writefile(a:buffer, l:lines, l:filename)

    return 'verilog_lint '
    \   . ale#Var(a:buffer, 'verilog_verible_options') .' '
    \   . ale#Escape(l:filename)
endfunction

function! ale_linters#verilog#verible#Handle(buffer, lines) abort
    " Look for lines like the following.
    "
    " filename.sv:7:13: Explicitly define static or automatic lifetime for non-class functions [Style: function-task-explicit-lifetime] [explicit-function-lifetime]
    let l:pattern = '\([A-z0-9]\+\.sv\):\(\d\+\):\d\+:\(.*\s\+\)\([Style:.*\)'
    let l:output = []
    for l:match in ale#util#GetMatches(a:lines, l:pattern)
        let l:file = l:match[1]
        let l:line = l:match[2]
        let l:text = l:match[3]
        let l:type = l:match[4]

        if l:file =~# '_verible_linted.sv'
            call add(l:output, {
            \   'lnum': l:line,
            \   'text': l:text,
            \   'type': l:type,
            \})
        endif
    endfor

    return l:output
endfunction

call ale#linter#Define('verilog', {
\   'name': 'verible',
\   'output_stream': 'stdout',
\   'executable': 'verible-verilog-lint',
\   'command': function('ale_linters#verilog#verible#GetCommand'),
\   'callback': 'ale_linters#verilog#verible#Handle',
\   'read_buffer': 0,
\})
```

Sample verily file ``encoder.v``:
```verilog
//-----------------------------------------------------
// Design Name : encoder_using_case
// File Name   : encoder_using_case.v
// Function    : Encoder using Case
// Coder       : Deepak Kumar Tala
//-----------------------------------------------------
module encoder_using_case(
binary_out , //  4 bit binary Output
encoder_in , //  16-bit Input
enable       //  Enable for the encoder
);
output [3:0] binary_out  ;
input  enable ; 
input [15:0] encoder_in ; 
     
reg [3:0] binary_out ;
      
always @ (enable or encoder_in)
begin
  binary_out = 0;
  if (enable) begin
    case (encoder_in) 
      16'h0002 : binary_out = 1; 
      16'h0004 : binary_out = 2; 
      16'h0008 : binary_out = 3; 
      16'h0010 : binary_out = 4;
      16'h0020 : binary_out = 5; 
      16'h0040 : binary_out = 6; 
      16'h0080 : binary_out = 7; 
      16'h0100 : binary_out = 8;
      16'h0200 : binary_out = 9;
      16'h0400 : binary_out = 10; 
      16'h0800 : binary_out = 11; 
      16'h1000 : binary_out = 12; 
      16'h2000 : binary_out = 13; 
      16'h4000 : binary_out = 14; 
      16'h8000 : binary_out = 15; 
   endcase
  end
end

endmodule
```

Error:
```
Error detected while processing /Users/reportaman/.vim/plugged/ale/autoload/ale/engine.vim:
line  574:
E127: Cannot redefine function <SNR>140_RunIfExecutable: It is in use
line  589:
E127: Cannot redefine function <SNR>140_RunLinter: It is in use
line  636:
E127: Cannot redefine function <SNR>140_GetLintFileValues: It is in use
line  677:
E127: Cannot redefine function <SNR>140_RunLinters: It is in use
line  695:
E127: Cannot redefine function ale#engine#RunLinters: It is in use
Error detected while processing function ale#events#LintOnEnter[5]..ale#Queue[33]..<SNR>129_Lint[20]..ale#engine#RunLinters[4]..<SNR>140_GetLint
FileValues[27]..<lambda>8[1]..<SNR>140_RunLinters[12]..<SNR>140_RunLinter[6]..<SNR>140_RunIfExecutable[27]..ale#linter#GetCommand[3]..ale_linter
s#verilog#verible#GetCommand:
line    4:
E117: Unknown function: ale#engine#ManageFile
```

Doix, 2021-05-23:
@reportaman Looks like MangeFile has moved from ale#engine to ale#command... Maybe this will work:

```
if !exists('g:ale_verilog_verible_options')
    let g:ale_verilog_verible_options = ''
endif

function! ale_linters#verilog#verible#GetCommand(buffer) abort
    let l:filename = tempname() . '_verible_linted.sv'

    " Create a special filename, so we can detect it in the handler.
    call ale#command#ManageFile(a:buffer, l:filename)
    let l:lines = getbufline(a:buffer, 1, '$')
    call ale#util#Writefile(a:buffer, l:lines, l:filename)

    return 'verilog_lint '
    \   . ale#Var(a:buffer, 'verilog_verible_options') .' '
    \   . ale#Escape(l:filename)
endfunction

function! ale_linters#verilog#verible#Handle(buffer, lines) abort
    " Look for lines like the following.
    "
    " filename.sv:7:13: Explicitly define static or automatic lifetime for non-class functions [Style: function-task-explicit-lifetime] [explicit-function-lifetime]
    let l:pattern = '\([A-z0-9]\+\.sv\):\(\d\+\):\d\+:\(.*\s\+\)\([Style:.*\)'
    let l:output = []
    for l:match in ale#util#GetMatches(a:lines, l:pattern)
        let l:file = l:match[1]
        let l:line = l:match[2]
        let l:text = l:match[3]
        let l:type = l:match[4]

        if l:file =~# '_verible_linted.sv'
            call add(l:output, {
            \   'lnum': l:line,
            \   'text': l:text,
            \   'type': l:type,
            \})
        endif
    endfor

    return l:output
endfunction

call ale#linter#Define('verilog', {
\   'name': 'verible',
\   'output_stream': 'stdout',
\   'executable': 'verible-verilog-lint',
\   'command': function('ale_linters#verilog#verible#GetCommand'),
\   'callback': 'ale_linters#verilog#verible#Handle',
\   'read_buffer': 0,
\})
```

That might get you to the next error. Have you tried ALE without the verible stuff, does it work with python?

reportaman, 2021-05-23:
@Doix While there is no error now, this does not show any lint warnings/errors... As if there is no linting going on

reportaman, 2021-05-23:
@Doix Also, yes! I have been using ALE with pretty much every language I come across, it works well for me.

reportaman, 2021-05-23:
@Doix Here we go, I improved your code, and this is how it looks like now & it works! I was using it with .v verilog files, maybe that was the problem. Now both .v and .sv are supported.

```vim
if !exists('g:ale_verilog_verible_options')
    let g:ale_verilog_verible_options = ''
endif

function! ale_linters#verilog#verible#GetCommand(buffer) abort
    let l:filename = tempname() . '_verible_linted.sv'

    " Create a special filename, so we can detect it in the handler.
    call ale#command#ManageFile(a:buffer, l:filename)
    let l:lines = getbufline(a:buffer, 1, '$')
    call ale#util#Writefile(a:buffer, l:lines, l:filename)

    return 'verible-verilog-lint '
    \   . ale#Var(a:buffer, 'verilog_verible_options') .' '
    \   . ale#Escape(l:filename)
endfunction

function! ale_linters#verilog#verible#Handle(buffer, lines) abort
    " Look for lines like the following.
    " Patterns:
    "   Style warnings
    "       filename.sv:7:13: Explicitly define static or automatic lifetime for non-class functions [Style: function-task-explicit-lifetime] [explicit-function-lifetime]
    "   Syntax errors
    "       filename.v:39:4: syntax error, rejected "endcase" (syntax-error).
    let l:pattern = '\([A-z0-9_\-/]\+\.v\|[A-z0-9_\-/]\+\.sv\):\(\d\+\):\d\+:\(.*\s\+\)\([Style:.*\|(syntax-error.*\)'
    let l:output = []
    for l:match in ale#util#GetMatches(a:lines, l:pattern)
        let l:file = l:match[1]
        let l:line = l:match[2]
        let l:text = l:match[3]
        let l:type = l:match[4]
        call add(l:output, {
        \   'lnum': l:line,
        \   'text': l:text,
        \   'type': l:type,
        \})
    endfor

    return l:output
endfunction

call ale#linter#Define('verilog', {
\   'name': 'verible',
\   'output_stream': 'stdout',
\   'executable': 'verible-verilog-lint',
\   'command': function('ale_linters#verilog#verible#GetCommand'),
\   'callback': 'ale_linters#verilog#verible#Handle',
\   'read_buffer': 0,
\})
```

I use above with this setting in .vimrc:

```vim
let g:ale_linters = {
\   'verilog': ['verible']
\}
```

I just need to now differentiate between different patterns of syntax errors, or style warnings, etc.

reportaman, 2021-05-23:
@Doix Updated the code above to differentiate between style warnings, and syntax errors.

rdburns, 2021-09-19:
I have verible lint working in VS Code on save, without needing to integrate into a specific SystemVerilog extension.
It isn't running as you type, only on saving the file.

Based off instructions about [integrating a custom local linter here](https://stackoverflow.com/questions/65793046/is-it-possible-to-define-custom-local-linters-in-vscode), 

1) Install [MultiCommand ](https://marketplace.visualstudio.com/items?itemName=ryuta46.multi-command) extension

2) Create a VSCode Task for running Verible, by adding to tasks.json (at user or workspace level as you desire), within the `"tasks": [ ]` list:
(Don't forget to alter to match path to your rules file. Note that you could use the `${workspaceFolder}` variable there if you need a rules file to be workspace specific.)

``` json
      {
            "label": "Lint Verible",
            "type": "shell",
            "command": "verible-verilog-lint",
            "args": [
                "${relativeFile}",
                "--rules_config=/home/<path_to_your_rules_file>"
            ],
            "options": {
                "cwd": "${workspaceFolder}"
            },
            "presentation": {
                "clear": true,
                "reveal": "never",
                "echo": true
            },
            "problemMatcher": [
                {
                    "owner": "Verible",
                    "fileLocation": [
                        "relative",
                        "${workspaceFolder}"
                    ],
                    "severity": "error",
                    "pattern": {
                        "regexp": "^(.*):(\\d+):(\\d+):\\s+syntax error, (.*)$",
                        "file": 1,
                        "line": 2,
                        "column": 3,
                        "message": 4
                    }
                },
                {
                    "owner": "Verible",
                    "fileLocation": [
                        "relative",
                        "${workspaceFolder}"
                    ],
                    "severity": "info",
                    "pattern": {
                        "regexp": "^(.*):(\\d+):(\\d+):\\s+(.*)$",
                        "file": 1,
                        "line": 2,
                        "column": 3,
                        "message": 4
                    }
                }
            ]
        }

```
Note that this will pickup actual compile errors as 'Problem' error, and will assign linter issues as Infos, You can change the severity to warning or error if you like.  There are two matchers because verible lint outputs 'syntax error' if there is an error in the typical severity location, but doesn't output any particular severity string for actual linting rule violations.

Now you should have a task you can run manually in a SV file with command palette `Tasks: Run Task` and select Lint Verible. Make sure that works before proceeding.

If you want to lint manually, you could just assign a keybinding to this.

3) Create Multi-Command that combines saving and linting.

Add to your settings.json:

``` json
    "multiCommand.commands": [
        {
          "command": "multiCommand.lintingVerible",
          "interval": 500,
          "sequence": [
            "workbench.action.files.save",
            { "command": "workbench.action.tasks.runTask",
              "args": "Lint Verible"
            }
          ]
        }
      ],
```

You can make sure this works manually by being in your SV file, making some changes without saving, and running in the command pallete "Multi command: Execute Multi Command", choose `multicommand.LintingVerible`.  It should save the file *and* update the Problem pane. You may want to introduce a syntax error to make sure it's changing.

The interval is to make sure the file is saved. You may want to make it shorter, to make it snappier.

4) Override save file key binding for Verilog files to run this multicommand to save *and* lint by command palette `Preferences: Keyboard Shortcuts (JSON)`.

``` json
 {
        "key": "ctrl+x ctrl+s",
        "command": "multiCommand.lintingVerible",
        "when": "editorLangId == systemverilog"
    },
    {
        "key": "ctrl+x ctrl+s",
        "command": "workbench.action.files.save",
        "when": "editorLangId != systemverilog"
    },
```
Of course you may have your own preferred save-file shortcut. Also, your `editorLangId` may be different depending on which SystemVerilog extension you were using. I believe that if you change language, the string you would want is the grayed out hint one in the language selector.  

Note that it won't run when you open a new file, only when you save.

rdburns, 2021-09-19:
> @rdburns Thanks for the emacs tip!
> If you pass `-` as a file name to `verible-verilog-lint` that will read from stdin.
> Does that help you avoid the temporary file?

Thanks for this by the way, I did try it, but I couldn't get it to work, probably on the emacs side, and I just stuck with what I already had working.

hzeller, 2024-05-06:
I think with the [language server](https://github.com/chipsalliance/verible/tree/master/verilog/tools/ls#readme) a lot of the manual integration needed can be be considered fixed now. It allows linting and formatting directly in any editor that supports the language server protocol which is essentially any code editor be it emacs, vim or vscode.

So closing now; if there is anything left to do, please re-open and comment.

closed 2024-05-06 by hzeller