chipsalliance/verible #202: alternate rule for explicit parameter storage

msfschaffner, 2020-02-15:
While we do also require parameters/localparams to explicitly define a storage type in OpenTitan, there is one exception: "old-school" string parameters like this one
```systemverilog
parameter ARCHITECTURE = "FAST";
```
Unfortunately defining the storage type as `string` causes trouble with some tools. Hence, it would be great if we could enable a sort of "legacy" mode for this rule which allows string parameters.

msfschaffner, 2020-02-15:
b/149605532

fangism, 2020-02-15:
blocked by #133 

The way this would probably look (proposed) via command-line control is something like:

`--rules=explicit-parameter-storage=-string` meaning apply this rule except for when right-hand-side is a string-literal.

msfschaffner, 2020-02-18:
Yes, sgtm.

closed 2020-03-17 by hzeller

hzeller, 2020-03-17:
This is now a parameter to the `explicit-parameter-storage` rule.

With `--rules=explicit-parameter-storage=exempt_type:string`, this rule will not complain if the right-hand-side of the parameter assignment is a string literal.

msfschaffner, 2020-03-17:
Great! Thanks for adding that feature.

commit 9cf08258 (2020-03-17): Make explicit parameter storage type configurable to
optionally allow string-parameters without types.

Fixes #202

PiperOrigin-RevId: 301228048