chipsalliance/verible #203: Linter rules for parameters do not match multiple inline parameters

tgorochowik, 2020-02-18:
For example the following code:
```
module params;
  parameter int foO = 1, baR = 1;
endmodule
```

Executed like this:
```
verilog_lint -rules parameter-name-style ../params.sv
```

Produces:
```
params.sv:2:17: Non-type parameter names must be styled with UpperCamelCase or ALL_CAPS. [Style: constants] [parameter-name-style]
```

The second parameter (`baR`) is not matched.

tgorochowik, 2020-02-18:
@fangism what would be the preferred way to approach this? I guess we could either modify the rule code to search for multiple parameters after `kParamDeclaration`, or make sure each entry has its own `kParamDeclaration`.

This issue (and the chosen solution) also affects #162

fangism, 2020-02-18:
Good find.  I would add support for being able to extract multiple assignment-initializations from a single parameter declaration, and then extract and analyze the name from each assignment-initialization.

fangism, 2020-02-18:
b/149754819

commit 50fc5eec (2020-02-20): PR #207: Match multiple inline parameters in linter name style rule
Fixes: #203

GitHub PR https://github.com/google/verible/pull/207

Copybara import of the project:

  - 63c0f8fc6b94ecf5c3a36264ababfa98cb3bbf03 CST: Add a helper to get all tokens for a param declaration by Tomasz Gorochowik <tgorochowik@antmicro.com>
  - 0695fa5104fa7fbb439c6fb8533916d3b7d69436 Handle multiple inline params declarations in name style ... by Tomasz Gorochowik <tgorochowik@antmicro.com>

Closes #207

PiperOrigin-RevId: 296280275
closed 2020-02-20 by hzeller