chipsalliance/verible #206: [git users] Format only changed lines

fangism, 2020-02-18:
`verilog_format` supports a `--lines` feature which allows selective formatting over ranges of lines.  This is useful for leaving untouched code as-is to minimize formatter churn between commits.  For git, here's how I would accomplish this (in some wrapper script):

* `git add -u` to stage files into index
* `git diff -u` to compute differences relative to latest commit, write to temporary diff file
* For every changed file in the diff: convert the `+` changed lines into line ranges N-M,P-Q...
  * Invoke `verilog_format --inplace` and feed `--lines` ranges for every file
* The resulting (non-cached) workspace diff will show formatting changes, leaving user to decide whether or not to accept them (`git add -u` or discard).
* Bonus points: plug into some sort of `git format` command/hook

fangism, 2020-02-18:
b/149034770

fangism, 2020-02-18:
@msfschaffner would this be valuable for deployment on your team project?

fangism, 2020-02-19:
Some prior art:

https://github.com/llvm-mirror/clang/blob/master/tools/clang-format/clang-format-diff.py
https://gist.github.com/AtnNn/10d17408516ae408788484418740f18b
(personally, I'd replace perl with awk...)

msfschaffner, 2020-02-25:
Sorry for the late response. That looks like a very useful feature - especially if we decide to not bulk-format all files in one shot. Let me discuss this with my colleagues, but i think I may give this a try.

commit df23ee7b (2020-04-04): Incrementally format changed lines script for git users.

Automatically scans git-diff output to emit --lines for each file.
New files are formatted wholly.
Non-verilog files are skipped.

fixes #206

PiperOrigin-RevId: 304527530
closed 2020-04-04 by hzeller