chipsalliance/verible #216: Verilog formatter uses > 26 GiB of memory

mark-christiaens, 2020-02-23:
I did a quick test of the verilog formatter.  I fed it the attached file (from the OpenSPARCT2 project I often use as a guinea pig) and the formatter was using more and more memory until I killed it using 26 GiB.  
[cpu.v.gz](https://github.com/google/verible/files/4241548/cpu.v.gz)

snsokolov, 2020-02-23:
Thanks for the report!

David, here is the reduced test case:

```
module foo;
  bar bar(
      .a(a),
`ifndef BAZ
`endif
  );
endmodule
```

fangism, 2020-02-23:
Confirmed.

b/150085360

fangism, 2020-02-23:
I recently added a script to help mass-triage formatter issues:
https://github.com/google/verible/blob/master/verilog/tools/formatter/triage_formatter.sh

fangism, 2020-02-25:
Adding triage notes:

This test case fails similarly (no comma after `.a(a)`:

```systemverilog
module foo;
  bar bar(
      .a(a)
`ifndef BAZ
`endif
  );
endmodule
```

as does this one:

```systemverilog
module foo;
  bar bar(
`ifndef BAZ
`endif
  );
endmodule
```

This is the stacktrace in common with the hanging failures (sampling with timeout):

```
    @     0x555f7c787a07        128  std::__u::__put_character_sequence<>()
    @     0x555f7c67d204         64  verible::FormattedToken::FormattedText()
    @     0x555f7c65ec8b       1024  verilog::formatter::FormatVerilog()
    @     0x555f7c6588b6       1088  main
    @     0x7f08c8a00bbd        208  __libc_start_main
    @     0x555f7c666b69  (unknown)  _start
```

fangism, 2020-02-25:
Running `verilog_format --show_token_partition_tree` on the last example reveals more:

```
...
  { ([<auto>]) @{2}, policy: fit-else-expand
    { ([module foo ;]) }
  }
  { (>>[<auto>]) @{3}, policy: always-expand
    { (>>[<auto>]) @{3,0}, policy: fit-else-expand
      { (>>[bar bar (]) }
      { (>>>>>>[<auto>]) @{3,0,1}, policy: fit-else-expand
        { (>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
...
*** SIGTERM received by PID 137190 (TID 137190) from PID 137189; stack trace: ***
PC: @     0x7fb047b79350  (unknown)  __write
    @     0x56536d03f10c       1952  FailureSignalHandler()
    @     0x7fb047d079a0  1134336768  (unknown)
    @     0x7fb047bd649c         48  new_do_write
    @     0x7fb047bd7da3         32  _IO_do_write@@GLIBC_2.2.5
    @     0x7fb047bd747a         64  _IO_file_xsputn@@GLIBC_2.2.5
    @     0x7fb047bcc0f1         48  fwrite
    @     0x56536d0ee9e9        128  std::__u::__put_character_sequence<>()
    @     0x56536cfe3837        384  verible::UnwrappedLine::AsCode()
    @     0x56536cfe2f6c        128  verible::TokenPartitionTreePrinter::PrintTree()
    @     0x56536cfe2efa        128  verible::TokenPartitionTreePrinter::PrintTree()
    @     0x56536cfe2efa        128  verible::TokenPartitionTreePrinter::PrintTree()
    @     0x56536cfe2efa        128  verible::TokenPartitionTreePrinter::PrintTree()
    @     0x56536cfe2efa        128  verible::TokenPartitionTreePrinter::PrintTree()
    @     0x56536cfc562b       1024  verilog::formatter::FormatVerilog()
    @     0x56536cfbf8b6       1088  main
    @     0x7fb047b7bbbd        208  __libc_start_main
    @     0x56536cfcdb69  (unknown)  _start
```

fangism, 2020-02-25:
I added a sanity check here:
https://github.com/google/verible/blob/64a4e8f0a448a93ee845442ff7074402820fac08/common/formatting/unwrapped_line.cc#L56

for `indentation_spaces_` being in a "reasonable range" of [0, 100), and got:

```
F0225 15:48:28.960829  191224 unwrapped_line.cc:56] Check failed: indentation_spaces_ >= 0 (-4 vs. 0) 
```

which is repeatable, so this isn't a matter of uninitialized memory.

fangism, 2020-02-26:
The negative value comes from:

https://github.com/google/verible/blob/64a4e8f0a448a93ee845442ff7074402820fac08/verilog/formatting/tree_unwrapper.cc#L1004

from https://github.com/google/verible/commit/7eaae8394592859e1aae239db815208463224e90 (which is my bad)

fangism, 2020-02-26:
Fix is on the way.

commit ea9aa58f (2020-02-27): Fix negative indentation value, which caused formatter to hang.

Unindented lines like preprocessor directives should remain at 0 indentation
after un-indentation adjustments.

fixes #216

PiperOrigin-RevId: 297294930
closed 2020-02-27 by hzeller