chipsalliance/verible #217: Forbid implicit declarations

corco, 2020-02-23:
SystemVerilog allow implicit net declaration (language specification section 6.10). While useful for netlists, in traditional code it allows bugs to go undetected:

- A typo can declare an extra net instead of using intended one
- Forgetting to declare a net will implicitly declare it as 1 bit net, which may not be the desired type.

In both cases, the resulting code is legal and may be partly functional, making it hard to detect the issue.

I would like to see implicit net detected and reported in the linter.

fangism, 2020-02-24:
Couldn't agree more, would love to see this.  (Have a horror story about this kind of issue...)
Even though this borders on semantic/symbolic linting, this (use/def checking) is achievable with the style linter.  None of the current style lint rules do any symbol tracking/resolution yet, that would be the first step.

fangism, 2020-02-24:
b/138353038

rdburns, 2020-12-30:
There is a somewhat easy way to deal with this.

At the top of a module file, you should set
``` verilog
`default_nettype none
```
This will cause an undeclared net to trigger a real error in whatever tool the user is using. 

A caveat: once you do this, any i/o wire have to be declared explicitly (many people just write ```input somesig```, and that takes advantage of the default nettype being wire).

A 2nd caveat: The default nettype persists across files after a compiler reads it, so you should to set default_nettype back to wire at the end of the file, as a lot of code assumes it.

``` verilog
`default_nettype wire
```

Now all that said, I would love there to be a lint rule to check if default_nettype is set to none at the start of a file and wire at the end, and that would be a lot easier to implement than trying to verify nets are declared.

chili-chips-ba, 2026-10-24:
One more case for this rule: a `localparam` used before its declaration (IEEE 1800 clause 6.1). `verible-verilog-lint` passes it. Icarus rejects it, and Verilator misses it too.

```systemverilog
module dau;
  real p;
  initial begin
    p = 1.0 * N;
    $display("%f", p);
  end
  localparam int N = 8;
endmodule
```

Could the rule in #244 cover parameters as well as variables?