chipsalliance/verible #218: non-ANSI module ports need to be partitioned

fangism, 2020-02-26:
After fixing the #216 on the test case https://github.com/google/verible/files/4241548/cpu.v.gz, one of the next issues is that non-ANSI module ports aren't sufficiently partitioned.

Reduced test case:

```systemverilog
module m(FOO, BAR, BAZ);
endmodule
```

Currently, we get this partitioning:

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/formatting/b150261216.sv
Full token partition tree:
{ ([<auto>]) @{}, policy: always-expand
  { ([<auto>]) @{0}, policy: fit-else-expand
    { ([module m (]) }
    { (>>>>[<auto>]) @{0,1}, policy: fit-else-expand
      { (>>>>[FOO , BAR , BAZ]) }
    }
    { ([) ;]) }
  }
  { ([endmodule]) }
}
```

but we want something more like:

```
{ ([<auto>]) @{}, policy: always-expand
  { ([<auto>]) @{0}, policy: fit-else-expand
    { ([module m (]) }
    { (>>>>[<auto>]) @{0,1}, policy: fit-else-expand
      { (>>>>[FOO ,]) }
      { (>>>>[BAR ,]) }
      { (>>>>[BAZ]) }
    }
    { ([) ;]) }
  }
  { ([endmodule]) }
}
```

This is the current CST dump:

```
$ verilog_syntax --printtree ~/testing/verilog/formatting/b150261216.sv
Parse Tree: 
Node (tag: kDescriptionList) {
  Node (tag: kModuleDeclaration) {
    Node (tag: kModuleHeader) {
      (#"module" @0-6: "module")
      (#SymbolIdentifier @7-8: "m")
      Node (tag: kParenGroup) {
        (#'(' @8-9: "(")
        Node (tag: kPortDeclarationList) {
          Node (tag: kPort) {
            Node (tag: kPortReference) {
              Node (tag: kUnqualifiedId) {
                (#SymbolIdentifier @9-12: "FOO")
              }
            }
          }
          (#',' @12-13: ",")
          Node (tag: kPort) {
            Node (tag: kPortReference) {
              Node (tag: kUnqualifiedId) {
                (#SymbolIdentifier @14-17: "BAR")
              }
            }
          }
          (#',' @17-18: ",")
          Node (tag: kPort) {
            Node (tag: kPortReference) {
              Node (tag: kUnqualifiedId) {
                (#SymbolIdentifier @19-22: "BAZ")
              }
            }
          }
        }
        (#')' @22-23: ")")
      }
      (#';' @23-24: ";")
    }
    Node (tag: kModuleItemList) {
    }
    (#"endmodule" @25-34: "endmodule")
  }
}
```

We probably just need handle the `kPort` case in 
https://github.com/google/verible/blob/master/verilog/formatting/tree_unwrapper.cc

fangism, 2020-02-26:
@mark-christiaens FYI, this affects your OpenSparc cpu.v test case.

fangism, 2020-02-26:
b/150261216

fangism, 2020-02-26:
@mark-christiaens Would you or someone from your team like to take a stab at this one?  It is among the easiest class of formatter issues to fix.  You have right-of-first-refusal.  :)

fangism, 2020-03-04:
I believe pull #224 covers this.

ldalek-antmicro, 2020-03-04:
Yes. It does.
Input:
```
module m(FOO, BAR, BAZ);endmodule
```
Output:
```
module m (
    FOO,
    BAR,
    BAZ
);
endmodule
```

```
Full token partition tree:
{ ([<auto>]) @{}, policy: always-expand
  { ([<auto>]) @{0}, policy: fit-else-expand
    { ([module m (]) }
    { (>>>>[<auto>]) @{0,1}, policy: always-expand
      { (>>>>[FOO ,]) }
      { (>>>>[BAR ,]) }
      { (>>>>[BAZ]) }
    }
    { ([) ;]) }
  }
  { ([endmodule]) }
}
```

ldalek-antmicro, 2020-03-04:
PR #224 merged. I guess we can close this issue, @fangism ?

fangism, 2020-03-04:
Indeed!

closed 2020-03-04 by fangism