chipsalliance/verible #230: How to use CST in other tools/languages

markusdd, 2020-03-14:
Hi,

thanks for driving this project, this will be very useful!

My question is simple (and maybe a feature request):

I want to sue the CST data to process it further in Python. We got some tooling for documentation and driver generation, where I wish to extract some module info like name, ports, portwidths, parameters etc.

I could of course use --printtree and read it in as text into python, parse it again in some way and be done with it. But it seems kind of silly to write a parser for a parser output. Is there any 'direct' way how to get this cst into Python? Is it a standardized output which I can access through some library to e.g. get a large dictionary structure which I can browse through or is it some proprietary format invented by this project.

Some guidance would be appreciated.

fangism, 2020-03-14:
Hi there.  There is "nothing special" about the CST, it is just a generic LISP structure (tree) of nodes and terminals, and Verilog-specific enums, so in theory it should be easy to export to any foreign language like Python (as a recursive list-of-lists/elements, or tuples/namedtuples) or LISP/Scheme, and tools like SWIG might help.  At the moment, the interface wrapping and object conversion doesn't exist, so we could use some help there.

Note too that the way the CST is built by verilog.y is fragile and prone to change.  Most of the CST construction is done by generic calls to `MakeTaggedNode` which is like a tuple with a label.  You'll find today that there are some inconsistencies in the way each node is built, but we've been taking the lazy approach of fix-as-needed and gradually adding more `Make*` functions to the verilog/CST/* library.  The other ongoing work is creating corresponding named accessor functions to descend through subtrees, as opposed to `get-element[i].get-element[j]...` direct positional access, the purpose being abstracting away the CST structural details when trying to query access information about specific nodes.  Writing those accessor functions today with unit-tests is a little tedious and manual -- you certainly would not want to replicate that effort in multiple languages (C++, Python, LISP).  It would make most sense to specify structural information independent of language, and emit equivalent code in multiple languages (maybe even including the unit-tests).  

Does this give you a better picture of where things stand?

markusdd, 2020-03-14:
Yes, that is insightful. Thanks very much.

I'm currently playing with Pythons anytree library and I've managed to write a relatively compact parser
for the current CST output. I'm now trying to pull out ports, parameters and instances.
On the lower level where it gets to the content this ends up being some rather nasty string manipulation at the moment.

So yes, what is actually needed is a 'structured way' of getting the data out of verible. Towards Python the obvious choice would be a combination of dictionaries and lists/tuples/sets .

I think this is anyway needed if this is going to be a base for e.g. editor plugins which wish to provide a object/module outline, hierarchy overview etc.

fangism, 2020-03-14:
More notes:
The CST tree dump was intended for human-eyes and may not have a stable form, and you might filter it down into an easier-to-parse form, for now.  Also know that the CST dump is abbreviated, in that it doesn't show you where `nullptrs` are, so the textual representation of `(node None foo None)` appears the same as `(node foo)`.  We could add a printing mode to show the nullptrs pretty easily.

Since you mentioned outline hierarchy viewing, you might be interested in #185 as well.

markusdd, 2020-03-14:
What crossed my mind is: Instead of providing expensive-to-implement language interfaces a standardized output like JSON would probably work well and should be easy to get with what you already have currently. JSON can be read by existing well-tested libs in almost any language out there.
Anotehr option is XML. In fact there is ipXact, a standard by the spirit consortium, which is basically the indsutry standard for HDL circuit descriptions for IP vendor and exchange purposes.

fangism, 2020-03-15:
I'd welcome a JSON or XML emitter.  Anything that depends on a proper schema will take more effort (and probably a proper AST).  Also note that the CST contains preprocessor directives, something that would need to be handled before targeting a standard like ipXact.

markusdd, 2020-03-15:
I currently lack the bandwidth to go into a full solution for this (and I'm no CPP crack either) but I happily share what I've started so far so others might use it:

```python
import re

import anytree as t

with open("path/to/cst/file/emitted/by/verilog_syntax --printtree", 'r') as f:
    content = f.readlines()

hier = []
nodes = []

tag_re = re.compile("(?<=Node\s\(tag:\s)[^\)]*")  # get the Node tag
info_re = re.compile("(?<=\(#)[^(\)\n)]*")  # get the content of Nodes
pos_re = re.compile("\s@\d*-\d*:\s")  # used to cut out the @xxxx-yyyy: from the content

for line in content[2:]:
    parent = None
    if hier:  # if hierarchy is empty we are the top node and have no parent
        parent = hier[-1]
    if line.lstrip().startswith("Node"):
        nodes.append(t.Node(tag_re.search(line).group(0), parent=parent, content=[]))
        hier.append(nodes[-1])  # new node, so we extend the hierarchy stack
    if line.lstrip().startswith("}"):  # closing node section, ascend in hierarchy
        hier.pop()
    if line.lstrip().startswith("(#"):  # content for the current hierarchy
        match = info_re.search(line).group(0)
        match = pos_re.sub("", match)
        nodes[-1].content.append(match)

assert not hier, "ERROR: Hierarchy stack is not empty, error while parsing!"

for pre, fill, node in t.RenderTree(nodes[0]):
    print("%s%s(%s)" % (pre, node.name, node.content))

```
This is in no way pretty, super generic or anything, but it helped me getting a python tree object (anytree is really a cool lib) which I can traverse and search for what I need.
For my private pupose I then have a few functions which pull out ports, params and instances and converts them in a format I need for further use.

fangism, 2020-08-18:
@markusdd Head's up.  With a19660768f29672f23355a78eaaea2f6e05c9268, the tree dump added a few more annotations, and may require small changes to your tree-dump parser.

fangism, 2020-11-20:
@imphil @tgorochowik in you're interested in exporting Verible's syntax trees and related data to another language.

It should be pretty easy to emit python syntax for the CST directly from C++ as well.  (You'll probably want node/leaf enumerations as well.)

fangism, 2020-11-23:
Simple idea: make a variant of `--printtree` that writes out the tree in python object notation:

```python
[SVNode.kModuleDeclaration, [SVNode.kModuleHeader, [...]]
```

Also provide an enum class with those constants, as generated from verilog_nonterminals.h.:

```python
class SVNode:
  """Syntax tree node enumerations generated from verilog_nonterminals.h."""
  kModuleDeclaration = ...
  kModuleHeader = ...
```

For the tree leaves, we have a few options: emit strings or byte-offset pairs to denote substring ranges within the larger source text.

In python, read in these forms and `eval()`.

From there, users can do whatever, with the caveat that the CST structure is not yet stable.

fangism, 2020-12-04:
Thanks for working on this @tgorochowik @mglb

tgorochowik, 2021-12-09:
I think this can be closed, the python helpers are fully usable

closed 2021-12-09 by tgorochowik