chipsalliance/verible #236: Linter configuration file - upward searching

tgorochowik, 2020-03-20:
Now that the initial linter configuration file support is merged (#229), we can work on adding support for searching for the file.

After some research I think it would be nice to mimic what [yapf](https://github.com/google/yapf) does.

Here is the code responsible for searching:
https://github.com/google/yapf/blob/b306e98387b61a6adaeda4a9ef17c02e4184f559/yapf/yapflib/file_resources.py#L67-L109

To sum up:
* define a default file name to be searched for (i.e. `.rules.verible_lint`)
* look for that file starting in the directory of the linted source file
* look up the directories until the file is found
* it's okay if nothing is found
* optional: if not found look for a default global in `/etc/` or `~/.config/verible`

Additional configuration flags:
* `--rule_config` - defaults to empty, if specified the configuration is read from this particular file, the searching mechanism is not used
* `--no_rule_config` - defaults to false, if enabled the searching mechanism is not used and no configuration file will be used
* The two flags above could in theory be combined into one, but I think it's cleaner with two flags

I'm going to start working on the implementation described above. I am opening this issue earlier that the PR to let you guys add some comments before it's done if you done if you done agree with something.

fangism, 2020-03-20:
Sounds like: "stop-at-the-first-config-file-found" in the order you described.

The two flags you describe still sound a little confusing.  Also boolean flags should not start with `--no` otherwise the flags library will provide a negation that looks like `--nono...`.
If you used a single flag to control, is there some pre-existing convention where a `*` prefix before the file argument signals to search?  e.g. `--rule_config=file` does not search, `--rule_config=*file` searches upward?  I'd be ok with `--rule_config=` (blank) disabling the mechanism altogether.

Thoughts from others?  @hzeller ?  @msfschaffner ?

tgorochowik, 2020-03-20:
> Sounds like: "stop-at-the-first-config-file-found" in the order you described.

Yes, that was the idea. Would you rather it combined all available files and combined them? That should be doable too.

> The two flags you describe still sound a little confusing. Also boolean flags should not start with `--no` otherwise the flags library will provide a negation that looks like `--nono...`.
> If you [...]

Ok, thanks for your feedback, I'll keep your remarks in mind

fangism, 2020-03-20:
> Yes, that was the idea. Would you rather it combined all available files and combined them? That should be doable too.
>
>
> Ok, thanks for your feedback, I'll keep your remarks in mind

I'm ok with starting simpler.  Imagine debugging some unexpected configuration that comes from multiple sources, could have unintended surprises.

commit d47b94c6 (2020-05-21): PR #237: Implement upward file searching of linter configuration
Initial implementation, this feature probably still needs some discussion.

For now it mostly follows what was described in #236, taking the initial comments into account.

Fixes #236.

GitHub PR https://github.com/google/verible/pull/237

Copybara import of the project:

  - 320c50895414f8b6dd14a76870d5318b39e162c8 Implement upward file searching of linter configuration by Tomasz Gorochowik <tgorochowik@antmicro.com>

Closes #237

PiperOrigin-RevId: 312583129
closed 2020-05-21 by hzeller