chipsalliance/verible #242: Lexer failure on unexpected EOF

pmaupin, 2020-03-25:
For example, if a file ends with the line:

```
`define THE_ANSWER 42
```
and there is no terminating line-feed, the entire file is discarded by the lexer.

fangism, 2020-03-25:
From the LRM 22.5.1: 
"The macro text can be any arbitrary text specified on the same line as the text macro name. If more than one line is necessary to specify the text, the newline character shall be preceded by a backslash ( \ ). _The first newline character not preceded by a backslash shall end the macro text_."

My interpretation is that without the terminal newline, it is considered malformed.  Of course, in practice, tools are often written to be more forgiving about missing newlines (a POSIX file requirement).
While I think the tools is correct, it could also be made more forgiving.

fangism, 2020-03-25:
Speaking of POSIX, newline termination of a file is checked by a lint rule:
https://github.com/google/verible/blob/master/verilog/analysis/checkers/posix_eof_rule.cc

But if the lexer dies and prevents further analysis, then you might not see that diagnostic.

pmaupin, 2020-03-25:
I don't disagree with any of your comments, but my use-case is to analyze a large pre-existing codebase, with ancient files, that works fine with tools from Cadence, Mentor Graphics, Synopsys, etc.

I don't necessarily have the ability to push back changes, even for things that are obviously wrong, unless those obvious wrongnesses wiill obviously propagate all the way to the streamed-out chip, so it would be great if I didn't need to make intermediate changes to files before verible will process them.

This use-case is obviously different than the linting one, where the end result is almost always to go change the offending source code to make it less offensive, but I don't think I'm the only one with this sort of problem.

fangism, 2020-03-25:
Understood, working with legacy/foreign code is important.

To handle this particular test case in your original post, we just need to modify the lexer.
Here is the excerpt that consumes macro definition bodies:

https://github.com/google/verible/blob/a9d1b6412820264dc3c279d4636e98c1a761545c/verilog/parser/verilog.lex#L1198

`DiscontinuedLine` is intended to grab the last newline terminated line of the macro definition body.  Currently this doesn't gracefully handle the lack of line terminator before EOF.  
**Assignment:** make it work, and add/update test cases in verilog_lexer_unittest.cc.

fangism, 2020-07-31:
Happen to have a lexer fix for this coming soon.

commit c1dbc80d (2020-08-05): Improve handling of unexpected EOFs in the lexer.

The important thing is to not call yylex() after an EOF was encountered.

This allows unterminated comments (and similar) to gracefully return
accumulated text, and possible signal returning a different token type.
For comments, we permissively treat a missing \n as if it were implicitly there.

This also fixes the case where a macro definition body doesn't properly end
with a \n.

#PermissivelyNotPOSIX

fixes #242

PiperOrigin-RevId: 324287279
closed 2020-08-05 by hzeller