chipsalliance/verible #249: Long macro arg expression not wrapped correctly

fangism, 2020-03-29:
Example formatter output looks incorrect:

```systemverilog
module w;
  `mXFb1ZwMvcO0k0Z53roH(So7c6mZurEIyyzh22cTkINfBB5qPO2Dl,
      DBEWJ8y8HfFuJW2E ==
          ({1'b0, ZAstNEKhAcsWSc0D3} + ((iQJAKroCn0kbWqa && AArit5fvJyzgFXy) ? gPfWeyYJ7 : '0)),
                        XY7, GYp)
endmodule
```

The macro argument expression that starts with `DBEWJ8y8HfFuJW2E` should be aligned with the first macro argument, which is appended after `(`.

fangism, 2020-03-29:
b/152706685

fangism, 2020-03-30:
Another example:

```systemverilog
task t;
`uvm_info(`gfn, $sformatf("x", yyyyyy.ggggggggggggggggg, zzzzz.hhhhhhhhhh(),  aaaaaaaaaa.bbbbbbbb(ccccccc)), UVM_LOW)       
endtask
```

formats to:

```systemverilog
task t;
  `uvm_info(`gfn,
      $sformatf("x", yyyyyy.ggggggggggggggggg, zzzzz.hhhhhhhhhh(), aaaaaaaaaa.bbbbbbbb(ccccccc)),
            UVM_LOW)
endtask
```

`$sformat` should be aligned with `` `gfn ``.

Ciprian167, 2020-07-18:
I couldn't recreate any of the scenarios. 

For the first example formatter outputs:
```systemverilog
module w;
  `mXFb1ZwMvcO0k0Z53roH(So7c6mZurEIyyzh22cTkINfBB5qPO2Dl,
                        DBEWJ8y8HfFuJW2E == ({1'b0, ZAstNEKhAcsWSc0D3} +
                                             ((iQJAKroCn0kbWqa && AArit5fvJyzgFXy) ? gPfWeyYJ7 : '0)
                            ),
                        XY7, GYp)
endmodule
```
`DBEWJ8y8HfFuJW2E` is aligned with `So7c6mZurEIyyzh22cTkINfBB5qPO2Dl`. But something feels wrong about the way `({1'b0, ZAstNEKhAcsWSc0D3} + ((iQJAKroCn0kbWqa && AArit5fvJyzgFXy) ? gPfWeyYJ7 : '0)),` got wrapped.

While for the second example:
```systemverilog
task t;
  `uvm_info(`gfn,
            $sformatf("x", yyyyyy.ggggggggggggggggg, zzzzz.hhhhhhhhhh(),
                      aaaaaaaaaa.bbbbbbbb(ccccccc)),
            UVM_LOW)
endtask
```
Which I think is OK.

fangism, 2020-07-20:
Confirming your recent observations.
Agree that the placement of the lone `),` is odd, but is a separate issue from the original focus.
(This is a matter of penalty tuning.)

I'll close this one as fixed.

closed 2020-07-20 by fangism