chipsalliance/verible #251: Use OpenTitan dv_macros.svh as a testcase for macro handling

msfschaffner, 2020-04-01:
I believe [this file](https://github.com/lowRISC/opentitan/blob/14ffb82597f9c29c254d36e5bdb40de5dfb6ee3f/hw/dv/sv/dv_utils/dv_macros.svh#L1-L288) from the OpenTitan codebase could be a good testcase for macro related issues. We already had previous issues with this file (#44), but it still generates syntax errors when passed through the Verible formatter.

Currently I get:
```
Error lex/parsing-ing formatted output.  Please file a bug.
First error: token: "if" at 36:5
```

When processed with `verilog_format --inplace hw/dv/sv/dv_utils/dv_macros.svh `

msfschaffner, 2020-04-01:
b/152918224

fangism, 2020-04-01:
Just a clarification, the formatter accepted the input, it's just that it failed its own output consistency check.   Need to take a closer look (when I'm less tired).

fangism, 2020-04-03:
Somewhat reduced test case:

```systemverilog
`ifndef FOO
  `define FOO(EXT_, BASE_, MSG_="", SEV_=fatal, ID_=`gfn) \
    if (!$cast(EXT_, BASE_)) begin \
      `uvm_``SEV_($sformatf( `"BASE_`", `"EXT_`", MSG_)) \
    end                           
`endif
```

```
$ verilog_format --nofailsafe_success ~/testing/verilog/formatting/OT-dv_macros.svh || echo FAIL
`ifndef FOO
  `define FOO(EXT_, BASE_, MSG_="", SEV_=fatal, ID_=`gfn) \
    if (!$cast(EXT_, BASE_)) begin \
      `uvm_``SEV_($sformatf( `"BASE_`", `"EXT_`", MSG_)) \
    end
`endif

Error lex/parsing-ing formatted output.  Please file a bug.
First error: token: "if" at 13:5
Problematic formatter output is:
<<EOF>>
FAIL
```

fangism, 2020-04-03:
Further reduced to:

```systemverilog
`define FOO(EXT_, BASE_, MSG_="", SEV_=fatal, ID_=`gfn) \ 
    if (!$cast(EXT_, BASE_)) begin \
      `uvm_``SEV_($sformatf( `"BASE_`", `"EXT_`")) \
    end
```

From this point it defies reduction.

fangism, 2020-04-03:
The formatter wasn't printing the formatted output on internal error, but when I fix that I see the problem is obvious:

```
Problematic formatter output is:
`define FOO(
    EXT_, BASE_, MSG_
        =
        ""
        ,
        SEV_
        =
        fatal
        ,
        ID_ = `gfn)
    \
    if (!$cast(EXT_, BASE_)) begin \
      `uvm_``SEV_($sformatf( `"BASE_`", `"EXT_`")) \
    end
<<EOF>>
```

fangism, 2020-04-04:
There are multiple problems afoot.

* There is insufficient partitioning the entire macro definition including the body is one long partition with no substructure.  At the very least, the macro formal arguments should be partitioned like other function/task formal arguments.
* Calculation of column position when searching/optimizing for line wrapping fails to account for the newline inside the macro definition body, which throws of the position tracking.  (This was a known issue.)
* Even though the token_annotator has marked `(*, PP_define_body)` as must-append (never break), a break is still occurring.

fangism, 2020-04-04:
Insufficient partitioning is filed as #252 .

fangism, 2020-04-04:
Column position tracking of multi-line tokens has been filed as #253 .