chipsalliance/verible #252: macro definition formal arguments need to be partitioned

fangism, 2020-04-04:
Test case:

```systemverilog
`define FOO(BAR1, BAR2, BAR3)
```

Currently partitioned as:

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/formatting/macros-defs.sv
Full token partition tree:
{ ([<auto>], policy: always-expand) @{}
  { ([`define FOO ( BAR1 , BAR2 , BAR3 ) ], policy: fit-else-expand) }
}
```

but should be more like (mock-up):

```
{ ([<auto>], policy: always-expand) @{}
  { ([<auto>], policy: append-fitting) @{}
    { ([`define FOO ( ], policy: fit-else-expand) }
    { (>>>>[BAR1 ,], policy: fit-else-expand) }
    { (>>>>[BAR2 ,], policy: fit-else-expand) }
    { (>>>>[BAR3 ,], policy: fit-else-expand) }
  }
}
```

Emphasis is on the partitioning of `BAR1, BAR2, BAR3`.

fangism, 2020-04-04:
b/153206439

fangism, 2020-04-04:
This blocks #251 .

fangism, 2020-04-04:
If it is easy to address in the same PR:

When there is a definition body, like this

```systemverilog
`define FOO(BAR1, BAR2, BAR3) blah_definition_body
```

Attach the definition body to the last partition so:

```
   ([BAR3 , blah_definition_body ])
```

wtatarski, 2020-06-29:
Hi @fangism 
I'm working on this issue and so far I managed to get the following result:
```
Full token partition tree:
{ ([<auto>], policy: always-expand) @{}
  { ([<auto>], policy: fit-else-expand) @{0}, (origin: "`define FOO...BAR2, BAR3)")
    { ([`define FOO (], policy: fit-else-expand) }
    { ([<auto>], policy: fit-else-expand) @{0,1}, (origin: "BAR1, BAR2, BAR3")
      { ([BAR1 ,], policy: fit-else-expand, (origin: "BAR1")) }
      { ([BAR2 ,], policy: fit-else-expand, (origin: "BAR2")) }
      { ([BAR3], policy: fit-else-expand, (origin: "BAR3")) }
    }
    { ([) ], policy: uninitialized) }
  }

```
by setting `kUntagged` with `PartitionPolicyEnum::kFitOnLineElseExpand`
```
--- a/verilog/formatting/tree_unwrapper.cc
+++ b/verilog/formatting/tree_unwrapper.cc
@@ -717,6 +717,7 @@ void TreeUnwrapper::SetIndentationsAndCreatePartitions(
     case NodeEnum::kAssumeClause:
     case NodeEnum::kAssertPropertyClause:
     case NodeEnum::kAssumePropertyClause:
+    case NodeEnum::kUntagged:
     case NodeEnum::kExpectPropertyClause: {
       VisitIndentedSection(node, 0, PartitionPolicyEnum::kFitOnLineElseExpand);
       break;
```
Is this correct direction for fixing this issue?

fangism, 2020-06-29:
I would not use `kUntagged`.  Something is seriously wrong in the syntax tree construction if you find a `kUntagged` node.

Instead, I would follow how function/task formal arguments work.  I think you should be able to add/move a few case statements in their vicinity to make them work the same way.

wtatarski, 2020-07-02:
It looks like tree is parsed in a wrong way. I think that kUntagged should be parsed into kMacroArgList.

```
`define FOO(BAR1)
```
parsed into
```
Parse Tree: 
Node (tag: kDescriptionList) {
  Node (tag: kPreprocessorDefine) {
    (#"`define" @0-7: "`define")
    (#PP_Identifier @8-11: "FOO")
    Node (tag: kParenGroup) {
      (#'(' @11-12: "(")
      Node (tag: kUntagged) {
        Node (tag: kUntagged) {
          (#PP_Identifier @12-16: "BAR1")
        }
      }
      (#')' @16-17: ")")
    }
    (#"<<`define-tokens>>" @17-17: "")
  }
}
```
So I suspect bug in verilog parser.

fangism, 2020-07-02:
The parsing isn't the problem, but the CST construction is.

You just need to edit here:

https://github.com/google/verible/blob/4d6929e0e059c7735b5023095a94120a3d833dd6/verilog/parser/verilog.y#L912

and replace `MakeNode($1)` with a `MakeTaggedNode(N::kMacroFormalParameterList, $1)` to establish a new type-tagged singleton list, and probably add a new NodeEnum in here:

https://github.com/google/verible/blob/master/verilog/CST/verilog_nonterminals.h

Then you can handle that enum case in verilog/formatting/tree_unwrapper.cc.

commit b458ff6d (2020-07-10): PR #356: macro definition partitioning
Fixes #252

GitHub PR https://github.com/google/verible/pull/356

Copybara import of the project:

  - f62f61a596c263b42351a261902f78551d64b391 Fixed partitioning of formal arguments in macro definition by Wojciech Tatarski <wtatarski@antmicro.com>

Closes #356

PiperOrigin-RevId: 320625972
closed 2020-07-10 by hzeller