chipsalliance/verible #254: bazel build broken: 'NoneType' value has no field or method 'short_path' in _gen_installer

fangism, 2020-04-05:
Our CI build is currently failing with:

```
$ bazel build ...
Starting local Bazel server and connecting to it...
ERROR: /tmp/verible_copybared/BUILD:15:1: in _gen_installer rule //:_install_gen: 
Traceback (most recent call last):
	File "/tmp/verible_copybared/BUILD", line 15
		_gen_installer(name = '_install_gen')
	File "/usr/local/google/home/fangism/.cache/bazel/_bazel_fangism/b057478e5059923c8cf2d338db1da6fa/external/com_github_google_rules_install/installer/def.bzl", line 33, in _gen_install_binary_impl
		sources.append(file.short_path)
	File "/usr/local/google/home/fangism/.cache/bazel/_bazel_fangism/b057478e5059923c8cf2d338db1da6fa/external/com_github_google_rules_install/installer/def.bzl", line 33, in sources.append
		file.short_path
'NoneType' value has no field or method 'short_path'
ERROR: Analysis of target '//:_install_gen' failed; build aborted: Analysis of target '//:_install_gen' failed; build aborted
```

I've filed this to https://github.com/bazelbuild/bazel/issues/11069

fangism, 2020-04-05:
b/153238985

hzeller, 2020-04-06:
Stop-gap addressed in https://github.com/google/verible/commit/a4c46f2862319919e2ca3e46f7a168f2a18e99de

Possibly this might be due to our wrong assumption that a filegroup is installable. Discussed in
https://github.com/google/bazel_rules_install/issues/24

fangism, 2020-04-28:
@hzeller Shall we undo the stop-gap?  now that root issue is fixed?

hzeller, 2020-04-28:
Good point, I hadn't looked at this recently. Let me try un-stop-gapping.

commit 8177dd80 (2020-04-28): Re-enable git verilog format with fixed installer

Fixes #254

PiperOrigin-RevId: 308901397
closed 2020-04-28 by hzeller