chipsalliance/verible #267: Syntax error on lines beggining with ','

JakubJatczak, 2020-04-14:
Running `verilog_format` with this reduced example:
```verilog
module mod_name (input bus_a, output bus_b
`ifdef USE_PG_PIN
    , input pg_pin
`endif
    );
endmodule 
```
fails with:
```
1.v:3:16: syntax error, rejected "," (https://github.com/google/verible).
1.v:6:1: syntax error, rejected "endmodule" (https://github.com/google/verible).
```

fangism, 2020-04-14:
Yes, this is a known limitation.  Verible only reads un-preprocessed files without proper preprocessing, and the grammar as implemented has limited support for where `` `ifdefs ``s may lie.

The formatter is also limited to working on files it can parse, subject to the parser's limitations.

Issue #228 is in the same class of issues, and mentions potential strategies in the future, such as composite parsing/formatting.

fangism, 2020-04-14:
I understand you may or may not have the liberty to rewrite the above example, but:

```systemverilog
module mod_name (input bus_a,
`ifdef USE_PG_PIN
    input pg_pin,
`endif
    output bus_b
    );
endmodule 
```

should be acceptable to Verible's parser/formatter.

fangism, 2020-04-15:
@pmaupin I think you meant this one is a dupe of #241 (but you mentioned in #268 ).

pmaupin, 2020-04-15:
Yes, fat-fingered URL on my tablet.  Removed extraneous comment in issue 268.  Thanks!