chipsalliance/verible #268: Syntax error on user defined primitives use

JakubJatczak, 2020-04-14:
Running `verilog_format` on below example:
```verilog
module mod_name ( X , A , B , vpwr , vgnd );
input vpwr, vgnd;
output  X ;
input  A , B ;

wire    X , A , B ;

reg csi_notifier;

specify
    if ((!B)) (A +=> X ) = (0:0:0,0:0:0);
    if ((B)) (A -=> X ) = (0:0:0,0:0:0);
    if ((!A)) (B +=> X ) = (0:0:0,0:0:0);
    if ((A)) (B -=> X ) = (0:0:0,0:0:0);
endspecify

xor  ( UDP_IN_X , B , A ) ;
USER_DEF_PRIM (UDP_OUT_X, UDP_IN_X, vpwr, vgnd) ;
buf  (X, UDP_OUT_X) ;
endmodule
```
fails with 
```
2.v:19:49: syntax error, rejected ";" (https://github.com/google/verible).
```
This example is valid in verilog-xl

fangism, 2020-04-14:
Known parser limitation: anonymous primitive instances aren't supported at this time, however, named primitive instances work as expected.

fangism, 2020-04-14:
b/36706412

pmaupin, 2020-04-15:
This is a dup of #241 so if you fix it, you can kill two issues with one patch :)

PR #1686 (2023-05-22): Added grammar for supporting anonymous intantiation to the YACC parser
This PR implements those grammar rules that allow the anonymous instantiation of gates or register variables (or user-defined primitives) to be considered valid grammar. For that changes were implemented to how references and calls are structured and some restrictions needed to be placed around function bodies

Fixes #268
closed 2023-05-22 by hzeller