chipsalliance/verible #270: Syntax error on '==' inside `$setuphold`

JakubJatczak, 2020-04-14:
Running `verilog_format` with below verilog file:
```verilog
module mod_name ( bus_a, bus_b, vcd, vchib, vda, vdio, vdio_q, vsa, vsd, vsio, vsio_q, vsw,
   en_h, hld_h_n, ref_sel, vrefgen_en, vinref );

   inout vsio_q;
   inout vchib;
   inout vda;
   inout vsa;
   inout vcd;
   inout vdio_q;
   inout vdio;
   inout vsw;
   inout vsio;
   inout vsd;
   wire pwr_good_active_mode = (vdio_q===1) && (vsd===0) && (vchib===1);
   wire pwr_good_hold_mode   = (vdio_q===1) && (vsd===0);

   inout bus_a;
   inout bus_b;
   input vrefgen_en;
   input  [4:0] ref_sel;
   input en_h;
   input hld_h_n;
   inout vinref;

   reg notif_ref_sel,  notif_vrefgen_en;
   reg notif_en_h;

   wire [4:0] ref_sel_buf;
   wire vrefgen_en_buf, hld_h_n_buf;

   wire [4:0] ref_sel_del;
   wire vrefgen_en_del, hld_h_n_del;

   parameter STARTUP_DELAY = 24000;
   integer startup_delay;
   initial startup_delay = STARTUP_DELAY;

   assign ref_sel_buf  = ref_sel_del;
   assign vrefgen_en_buf   = vrefgen_en_del;
   assign hld_h_n_buf  = hld_h_n_del;
   specify

       $width  (negedge hld_h_n,         (15.500:0:15.500));
       $width  (posedge hld_h_n,         (15.500:0:15.500));

        specparam tsetup = 5;
        specparam thold = 5;

       $setuphold (posedge en_h,    negedge hld_h_n,    tsetup, thold,  notif_en_h);
       $setuphold (negedge hld_h_n, posedge ref_sel[4],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[4]);
       $setuphold (negedge hld_h_n, negedge ref_sel[4],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[4]);
       $setuphold (negedge hld_h_n, posedge ref_sel[3],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[3]);
       $setuphold (negedge hld_h_n, negedge ref_sel[3],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[3]);
       $setuphold (negedge hld_h_n, posedge ref_sel[2],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[2]);
       $setuphold (negedge hld_h_n, negedge ref_sel[2],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[2]);
       $setuphold (negedge hld_h_n, posedge ref_sel[1],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[1]);
       $setuphold (negedge hld_h_n, negedge ref_sel[1],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[1]);
       $setuphold (negedge hld_h_n, posedge ref_sel[0],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[0]);
       $setuphold (negedge hld_h_n, negedge ref_sel[0],   tsetup, thold,  notif_ref_sel,   en_h==1'b1, en_h==1'b1, hld_h_n_del, ref_sel_del[0]);
       $setuphold (negedge hld_h_n, posedge vrefgen_en,   tsetup, thold,  notif_vrefgen_en,en_h==1'b1, en_h==1'b1, hld_h_n_del, vrefgen_en_del);
       $setuphold (negedge hld_h_n, negedge vrefgen_en,   tsetup, thold,  notif_vrefgen_en,en_h==1'b1, en_h==1'b1, hld_h_n_del, vrefgen_en_del);
    endspecify

   wire x_on_vinref = vrefgen_en_final === 1'bx || (vrefgen_en_final === 1'b1 && ^ref_sel_final === 1'bx);
   assign #(startup_delay,0) vinref = x_on_vinref === 1'b1 ? 1'bx : vrefgen_en_final;

endmodule
```
Fails with:
```
4.v:52:97: syntax error, rejected "==" (https://github.com/google/verible).
4.v:53:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:54:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:55:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:56:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:57:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:58:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:59:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:60:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:61:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:62:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:63:9: syntax error, rejected "$setuphold" (https://github.com/google/verible).
4.v:64:6: syntax error, rejected "endspecify" (https://github.com/google/verible).
```

fangism, 2020-04-14:
Confirmed.  Reduced test case (fails `verilog_syntax`):

```systemverilog
module m;
specify
       $setuphold (negedge hld_h_n, posedge ref_sel[4],
	       tsetup, thold,  notif_ref_sel,
	       en_h==1'b1, en_h==1'b1,
	       hld_h_n_del, ref_sel_del[4]);
endspecify
endmodule
```

fangism, 2020-04-14:
This looks fixable in verilog.y: in the rule for `spec_notifier`, I just need to allow expressions, not only references.