chipsalliance/verible #271: Syntax error on wire delay value

JakubJatczak, 2020-04-14:
Running `verilog_format` on below example:
```verilog
module mod_name (out1_vdd_hv, out1_vdio_hv, out2_vdd_hv, out2_vdio_hv, out3_vdd_hv, out3_vdio_hv, tie_lo_esd, vdd_present_vdio_hv, vdio_present_vdd_hv,
    vcd, vdd1, vdd2, vdio_q, vsa, vsd, vsio_q, in1_vdd_hv, in1_vdio_hv, in2_vdd_hv, in2_vdio_hv, in3_vdd_hv, in3_vdio_hv ,rst_por_hv_n
);

    output out1_vdd_hv;
    output out1_vdio_hv;
    output out2_vdd_hv;
    output out2_vdio_hv;
    output out3_vdd_hv;
    output out3_vdio_hv;
    output tie_lo_esd;
    output vdd_present_vdio_hv;
    output vdio_present_vdd_hv;

    inout vcd;
    inout vdd1;
    inout vdd2;
    inout vdio_q;
    inout vsa;
    inout vsd;
    inout vsio_q;
    input in1_vdd_hv;
    input in1_vdio_hv;
    input in2_vdd_hv;
    input in2_vdio_hv;
    input in3_vdd_hv;
    input in3_vdio_hv;
    input rst_por_hv_n;

parameter STARTUP_DELAY = 2000;
integer startup_delay;
initial startup_delay = STARTUP_DELAY;

wire pwr_good_vdio_present_vdd_hv     =  (vdd1===1) && (vsa===0) && (vsd===0);
wire x_on_vdio_present_vdd_hv =  pwr_good_vdio_present_vdd_hv===0 || ^rst_por_hv_n===1'bx || ^vdio_q===1'bx;

assign vdio_present_vdd_hv    = x_on_vdio_present_vdd_hv===1 ? 1'bx : ((rst_por_hv_n===0 && vcd===0 && vdio_q===0) ? 1'b0 : vdio_q);
assign  out1_vdd_hv    =   vdio_present_vdd_hv && in1_vdio_hv;
assign  out2_vdd_hv    =   vdio_present_vdd_hv && in2_vdio_hv;
assign  out3_vdd_hv    =   vdio_present_vdd_hv && in3_vdio_hv;

wire pwr_good_vdd_present_vdio_hv     =  (vdio_q===1) && (vsa===0) && (vsio_q===0);
wire x_on_vdd_present_vdio_hv     =  (pwr_good_vdd_present_vdio_hv===0) || (^vdd2===1'bx);
wire vdd_present_vdio_hv_buf      =   x_on_vdd_present_vdio_hv===1 ? 1'bx : vdd2;

assign #(startup_delay,startup_delay,0) vdd_present_vdio_hv   =   vdd_present_vdio_hv_buf;
assign  out1_vdio_hv   =    (vdd_present_vdio_hv && in1_vdd_hv);
assign  out2_vdio_hv   =    (vdd_present_vdio_hv && in2_vdd_hv);
assign  out3_vdio_hv   =    (vdd_present_vdio_hv && in3_vdd_hv);

assign tie_lo_esd = vsd;

reg dis_err_msgs;

initial
begin
  dis_err_msgs = 1'b1;
  `ifdef DIS_ERR_MSGS
  `else
    #1;
    dis_err_msgs = 1'b0;
  `endif
end

event event_warning_unreliable_output;

wire #(100) warning_unreliable_output = rst_por_hv_n===1 && vcd!==1 && vdio_q===0;

specify
  (in1_vdio_hv   => out1_vdd_hv) =  (0:0:0 , 0:0:0);
  (in2_vdio_hv   => out2_vdd_hv) =  (0:0:0 , 0:0:0);
  (in3_vdio_hv   => out3_vdd_hv) =  (0:0:0 , 0:0:0);

  (in1_vdd_hv    => out1_vdio_hv) =  (0:0:0 , 0:0:0);
  (in2_vdd_hv    => out2_vdio_hv) =  (0:0:0 , 0:0:0);
  (in3_vdd_hv    => out3_vdio_hv) =  (0:0:0 , 0:0:0);

  $width  (negedge rst_por_hv_n,        (0:0:0));
endspecify

endmodule

 ```
fails with :
```
5.v:67:6: syntax error, rejected "#" (https://github.com/google/verible).
```

fangism, 2020-04-14:
Confirmed.  Reduced test case:

```systemverilog
module m;
wire #(100) x = y;
endmodule
```

The LRM grammar allows this as:

net_declaration : net_type [...] data_type_or_implicit [delay3] list_of_net_decl_assignments `;`

fangism, 2022-03-25:
b/161138189

closed 2022-03-28 by hzeller