chipsalliance/verible #276: [unpacked-dimensions-range-ordering] rule should exclude module array instances

fangism, 2020-04-20:
The following test case:

```systemverilog
module test (input logic [1:0] blah);
endmodule

module tb;
test test_i[3:0] (
  .blah (2'b0)
);
endmodule
```

is caught by `verilog_lint` as "[unpacked-dimensions-range-ordering]".
We would like to exempt module instance arrays from this check.

fangism, 2020-04-20:
b/154451633

fangism, 2020-04-20:
Running `verilog_syntax --printtree` on that input reveals that we can use the syntax tree context to refine the finding.  The tree dump includes:

```
...
          Node (tag: kGateInstanceRegisterVariableList) {
            Node (tag: kGateInstance) {
              (#SymbolIdentifier @65-71: "test_i")
              Node (tag: kUnpackedDimensions) {
                Node (tag: kDeclarationDimensions) {
                  Node (tag: kDimensionRange) {
                    (#'[' @71-72: "[")
                    Node (tag: kExpression) {
                      Node (tag: kNumber) {
                        (#TK_DecNumber @72-73: "3")
                      }
                    }
                    (#':' @73-74: ":")
                    Node (tag: kExpression) {
                      Node (tag: kNumber) {
                        (#TK_DecNumber @74-75: "0")
                      }
                    }
                    (#']' @75-76: "]")
                  }
                }
              }
...
```

So the check for exemption should use `kGateInstance`.

Ciprian167, 2020-05-03:
If this is still free, I'd like to take a shot at it. I cannot self assign it yet.

fangism, 2020-05-03:
Thank you for offering, let me know if you have any questions.

Ciprian167, 2020-05-03:
I think this could be waived in UnpackedDimensionsRule::HandleSymbol by changing the return condition to something like:
``` C++
if (!ContextIsInsideUnpackedDimensions(context) || 
     context.IsInside(NodeEnum::kGateInstance)) return;
```
What do think?

fangism, 2020-05-03:
> I think this could be waived in UnpackedDimensionsRule::HandleSymbol by changing the return condition to something like:
> 
> ```c++
> if (!ContextIsInsideUnpackedDimensions(context) || 
>      context.IsInside(NodeEnum::kGateInstance)) return;
> ```
> 
> What do think?

I think that might do.  Suggest building up some test cases to verify.

commit ea120a2a (2020-05-06): PR #288: Fix for #276: unpacked dimensions rules doesn't apply to array instances

GitHub PR https://github.com/google/verible/pull/288

Copybara import of the project:

  - e792646178eeecdca3884f07813d788260819448 Fix for #276 by Ciprian A <ciprian.antoci@gmail.com>
  - 653dda65d387c2feaa3c44a1ac4730ca7244c8a5 Changes for "array of instances" test. by Ciprian A <ciprian.antoci@gmail.com>

Closes #288
fixes #276

PiperOrigin-RevId: 310236310
closed 2020-05-06 by hzeller