chipsalliance/verible #277: general task call args need partitioning

fangism, 2020-04-21:
This feels similar to issues filed in the past, but isn't working yet.

Consider the input:

```systemverilog
  task t;
        if (!$cast(ssssssssssssssss, vvvvvvvvvv.gggggggg(
                .ppppppp(ppppppp),
                .yyyyy("xxxxxxxxxxxxx")
        ))) begin
        end 
  endtask : t
```

This formats to:

```systemverilog
task t;
  if (!$cast(ssssssssssssssss, vvvvvvvvvv.gggggggg(
  .ppppppp(ppppppp),
  .yyyyy("xxxxxxxxxxxxx")
  ))) begin
  end
endtask : t
```

Diagnosing the token partition tree reveals:

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/formatting/if-cast-call-args.sv
Full token partition tree:
{ ([<auto>], policy: always-expand) @{}
  { ([<auto>], policy: fit-else-expand) @{0}
    { ([task t ;], policy: append-fitting-sub-partitions) }
    { (>>[<auto>], policy: always-expand) @{0,1}
      { (>>[if ( ! $cast ( ssssssssssssssss , vvvvvvvvvv . gggggggg (], policy: fit-else-expand) }
      { (>>[. ppppppp ( ppppppp ) ,], policy: fit-else-expand) }
      { (>>[. yyyyy ( "xxxxxxxxxxxxx" )], policy: fit-else-expand) }
      { (>>[) ) ) begin], policy: fit-else-expand) }
      { (>>[end], policy: always-expand) }
    }
    { ([endtask : t], policy: always-expand) }
  }
}
```

Note that we already have #240 for the method call args, which in the above example is inside a plain `$cast(...)` call args.  This issue shall focus on the direct call args of `$cast`.

@ldalek-antmicro can you investigate?

fangism, 2020-04-21:
b/152765818

fangism, 2020-04-21:
Smaller more focused example:

```systemverilog
  task t;
        if (!$cast(ssssssssssssssss, vvvvvvvvvv, gggggggg)) begin
        end
  endtask : t
```

`verilog_format --show_token_partition_tree`:

```
{ ([<auto>], policy: always-expand) @{}
  { ([<auto>], policy: fit-else-expand) @{1}
    { ([task t ;], policy: append-fitting-sub-partitions) }
    { (>>[<auto>], policy: always-expand) @{1,1}
      { (>>[if ( ! $cast ( ssssssssssssssss , vvvvvvvvvv , gggggggg ) ) begin], policy: fit-else-expand) }
      { (>>[end], policy: always-expand) }
    }
    { ([endtask : t], policy: always-expand) }
  }
}
```

ldalek-antmicro, 2020-08-16:
@fangism I did formatting of this example with latest Verible and it looks like this commit https://github.com/google/verible/commit/2edbccd4f72f536dbf8815541d7d8251183258da is this this issue
Input:
<pre>
  task t;
        if (!$cast(ssssssssssssssss, vvvvvvvvvv, gggggggg)) begin
        end
  endtask : t
</pre>
Output:
<pre>
task t;
  if (!$cast(ssssssssssssssss, vvvvvvvvvv, gggggggg)) begin
  end
endtask : t
</pre>
Partition tree:
<pre>
Full token partition tree:
{ ([<auto>], policy: always-expand) @{}
  { ([<auto>], policy: always-expand) @{0}, (origin: "task t;
   ...endtask : t")
    { ([task t ;], policy: fit-else-expand, (origin: "task t;")) }
    { (>>[<auto>], policy: always-expand) @{0,1}, (origin: "if (!$cast(...        end")
      { (>>[<auto>], policy: fit-else-expand) @{0,1,0}, (origin: "if (!$cast(... gggggggg))")
        { (>>[if ( ! $cast (], policy: fit-else-expand) }
        { (>>>>>>>>>>[<auto>], policy: fit-else-expand) @{0,1,0,1}, (origin: "sssssssssss...v, gggggggg")
          { (>>>>>>>>>>[ssssssssssssssss ,], policy: fit-else-expand, (origin: "ssssssssssssssss")) }
          { (>>>>>>>>>>[vvvvvvvvvv ,], policy: fit-else-expand, (origin: "vvvvvvvvvv")) }
          { (>>>>>>>>>>[gggggggg], policy: fit-else-expand, (origin: "gggggggg")) }
        }
        { (>>>>>>[) ) begin], policy: uninitialized) }
      }
      { (>>[end], policy: always-expand, (origin: "end")) }
    }
    { ([endtask : t], policy: always-expand, (origin: "endtask")) }
  }
}
</pre>

Can you confirm?

fangism, 2020-08-16:
Confirmed.  Can you verify if this test case is already covered in tree_unwrapper_test and formatter_test?
If not, could you add this?

commit 82a9c862 (2020-08-20): PR #408: Add tests related to general task partitioning
Fixes #277

GitHub PR https://github.com/google/verible/pull/408

Copybara import of the project:

  - a3007bf57e2743e74459b4c2118b4b967834a705 Add tests related to general task partitioning by Lukasz Dalek <ldalek@antmicro.com>

Closes #408

PiperOrigin-RevId: 327543049
closed 2020-08-20 by hzeller