chipsalliance/verible #279: Automatically generate and publish API documentation

mithro, 2020-04-27:
Verilog should publish documentation on ReadTheDocs like all good packages.

See https://docs.readthedocs.io/en/stable/ on how to get started.

The docs should be published at https://verible.readthedocs.io/

mithro, 2020-04-27:
@rw1nkler and @mgielda from @antmicro might also be able to give some tips and hints.
@daniellimws might also be interested in helping out here. He has been doing some great work for other SymbiFlow related projects.

mithro, 2020-04-27:
The idea at https://github.com/SymbiFlow/ideas/issues/49 could help with making this nice and easy.

fangism, 2020-04-27:
At this phase there isn't a formal API yet, nor any commitment to keep certain library interfaces stable.  Current focus is on a few end-point tools.