chipsalliance/verible #281: ';' after `uvm_* statements is unwanted

fangism, 2020-04-28:
We've resolved to be consistent on this subject.

All `` `uvm_* `` _statements_, shall not be terminated with `;`.

Task:

* Examine the syntax tree difference between macro-call statements with vs. without `;` (use `verilog_syntax --printtree`)
* Identify all macro-call statements that start with `` `uvm_ `` in a new `SyntaxTreeLintRule` subclass.
* Check if they terminate with a `;`, and report them.
* Write unit-test test cases.

fangism, 2020-04-28:
b/155128436

svenka3, 2020-04-28:
As a regular user and reveiwer of UVM code, I would like to point out that, though this rule is a good one, many user environments violate this one consistently. It is so hard to educate them and trainers across the globe have given up on this one (Cliff Cumming's words at DVCon US 2020). Also I have heard one EDA vendor arguing it is still a legal SV syntax (A null or something). Also most of the simulators accept this extra ;

Hence, my suggestion would be not to add this rule - most violated one it shall become.

Regards
Srini

fangism, 2020-04-28:
Hi there, this request came from a consensus reached by internal users, and doesn't necessarily mean it has to be imposed on everyone else -- just those who want it moving forward for consistency, and with less interest about fixing legacy code.  With ever-improving configurability in the style linter, you can have it your own way and opt-out.

fangism, 2020-04-28:
Also tools never get exhausted trying to educate users.  :)

Ciprian167, 2020-05-06:
@fangism If it's still free, I would like to take a look at it. Could you please assing it to me?

commit 5383335f (2020-06-02): PR #308: Initial implementation of what could be a solution to #281.
UvmMacroSemicolonRule checks if there are any unwanted ';' after a `uvm_* macro call.
Also, a new method has been added in CST/macro.h that makes rule implementation a bit cleaner.

GitHub PR https://github.com/google/verible/pull/308

Copybara import of the project:

  - 1975dbca0766ce991879dcf3e8deba7b0c658e72 Initial implementation of what could be a solution to #281. by Ciprian A <ciprian.antoci@gmail.com>
  - 77c1a18d1f4ea38cbb4afcf9e4fdc7ff000c4d0d Changing rule strategy and adding test improvements. by Ciprian A <cipri [... cut]
closed 2020-06-02 by hzeller