chipsalliance/verible #282: verilog_obfuscate incorrectly obfuscates certain system functions

jrrk2, 2020-04-28:
verilog_obfuscate does not recognise system functions such as $signed and $clog2. If you have such in your code you can apply this temporary workaround:

verilator -E unobfusc.sv | grep -v '^`line' | grep -v '^ $' | cat -s | verilog_obfuscate --save_map obfusc.map.tmp > /dev/null
sed -e 's=^\(signed\)\ [^\ ]*=\1\ \1=' -e 's=^\(clog2\)\ [^\ ]*=\1\ \1=' < obfusc.map.tmp > obfusc.map
verilator -E unobfusc.sv | grep -v '^`line' | grep -v '^ $' | cat -s | verilog_obfuscate --load_map obfusc.map  > obfusc.sv

The proper fix is for the tool to have a table of such functions I suppose.

fangism, 2020-04-28:
b/153886384

jrrk2, 2020-04-28:
Excuse my ignorance, is that a bug number? Is there a bug tracker somewhere ?

fangism, 2020-04-28:
Confirmed.  The present default mode aggressively mutates all `$foo` task calls (because my first use of the obfuscator was for formatting).

Another potential workaround:
Create an identity mapping of known standard system task calls, like `$display` and prepopulate them into a text file like so, as a identity identifier map (without the leading `$`):

```
display display
sformat sformat
...
```

and pass that into `verilog_obfuscate --load_map FILE`.

fangism, 2020-04-28:
> Excuse my ignorance, is that a bug number? Is there a bug tracker somewhere ?

That's only an internal cross-reference bug number, you may ignore.

fangism, 2020-04-28:
I would accept changing the default behavior to preserve all `$foo` task calls, if someone would like to update here (with unit-tests):

https://github.com/google/verible/blob/5cf83fe89203ad4f508771fb12cac9433c56adba/verilog/transform/obfuscate.cc#L52

commit 2c98e209 (2020-05-04): PR #283: System calls such as $display and $sformat fail if encrypted
This patch recognizes a SystemTFIdentifier and turns off encryption for that token. Otherwise the obfuscated Verilog cannot be compiled if it contains such calls.

GitHub PR https://github.com/google/verible/pull/283

Copybara import of the project:

  - 095fae5fa2bbb5734e7192aa840a388e0679081d System calls such as  and  fail if encrypted by Jonathan Kimmitt <jonathan.kimmitt@pcsemi.com>
  - 04a1b60c1dd80ac6654f7ec69aaed14653c90b5a Correct testcases for unencrypted  tasks (fixes #282) by Jonathan Kimmitt <jonathan.kimmitt@pc [... cut]
closed 2020-05-04 by hzeller