chipsalliance/verible #284: Generate suggested-fixes for style-lint findings with trivial remedies

fangism, 2020-04-28:
A subset of style lint rule findings point to issues that are trivially fixable and should be made easy to apply, e.g. in the form of patches.

e.g.
* add/fix end labels
* delete extraneous semicolons

The benefit to users is that they could apply these suggested changes with minimal effort, instead of manual editing.

Project goal: Enhance the style linter so that it can include suggested-changes (diffs/edits).
Outputs should be made available in a structured manner so that it could be easily plumbed into various interfaces of editors, or rendered into a patch set (which could them be interactively/batch applied).

fangism, 2020-06-02:
Minimum requirements:

Each `LintViolation` should also be able to take an array of "Edits" where each Edit contains the information for text deletions/insertions/substitutions, in the form of string_views and offsets.  (Note that string_views and offsets can be interchanged when the entire enclosing text is known from `TextStructureView::Content()`.) Offsets should reflect the positions of text *before* any other edits are made.

Applying the set of Edits can and should be left to another phase of handling, perhaps patch generation, or interactive handling.

fangism, 2020-10-06:
To help with this task, there is also now diff/patching libraries in:

* https://cs.opensource.google/verible/verible/+/master:common/strings/diff.h
* https://cs.opensource.google/verible/verible/+/master:common/strings/patch.h

With those, one has options to generate diffs of suggested edits, and also interactively ([y/n]?) apply them.

tgorochowik, 2021-01-22:
@fangism @hzeller

@mglb is currently looking at this issue. Would you mind adding him to the repo so he can assign himself?
The same goes for @pawelsag who is currently working on other issues in Verible