chipsalliance/verible #285: centos-6 and centos-7 releases fail due to missing devtoolset-6

hzeller, 2020-04-28:
Travis currently has issues compiling for these centos targets, thus all travis builds look like they failed (though the regular build and test is just fine)

https://travis-ci.org/github/google/verible/jobs/680765831#L1521
https://travis-ci.org/github/google/verible/jobs/680765832#L2054

These are referenced in https://github.com/google/verible/blob/master/Docker/docker-generate.sh - are there newer versions ?

hzeller, 2020-04-28:
Tim, you might be able to quickly see what is going on.

hzeller, 2020-05-07:
CentOS 6 and 7 are still consistently failing to build; if you're busy, I can have a look tonight @mithro

mithro, 2020-05-07:
I won't be able to look at this until next week at the earliest. Feel free to take a look

commit 0b0011b5 (2020-05-08): PR #296: Fix centos 6/7 build.
Change devtoolset for centos 6 and 7 for compilation to work. (to fix #285 )

GitHub PR https://github.com/google/verible/pull/296

Copybara import of the project:

  - e3bf1292490c8b2475fdff051eaa5ab500b37ad3 (testig) by Henner Zeller <h.zeller@acm.org>
  - 5da6e46a7c4f4c8da1f85f3c1a8dd64c516a3637 ... more references to devtoolset. by Henner Zeller <h.zeller@acm.org>

Closes #296
fixes #285

PiperOrigin-RevId: 310616568
closed 2020-05-08 by hzeller