chipsalliance/verible #287: Unwanted space before `#(` in actual type parameters

fangism, 2020-05-01:
Example formatter output of reduced test case:

```systemverilog
task t;
  a_pkg::foo #(x)::func();
endtask
```

Should be:

```systemverilog
task t;
  a_pkg::foo#(x)::func();
endtask
```

fangism, 2020-05-01:
b/155444434

fangism, 2020-05-01:
This can be fixed here:

https://github.com/google/verible/blob/8177dd80cff4ba2ae884451e6c5316edb2b4dbe7/verilog/formatting/token_annotator.cc#L450

by adding a case that checks for the right token `#`'s context.

`verilog_syntax --printtree` on the test case shows this excerpt:

```
      Node (tag: kStatement) {
        Node (tag: kReferenceCallBase) {
          Node (tag: kFunctionCall) {
            Node (tag: kLocalRoot) {
              Node (tag: kQualifiedId) {
                Node (tag: kUnqualifiedId) {
                  (#SymbolIdentifier @9-14: "a_pkg")
                }
                (#"::" @14-16: "::")
                Node (tag: kUnqualifiedId) {
                  (#SymbolIdentifier @16-19: "foo")
                  Node (tag: kActualParameterList) {
                    (#'#' @19-20: "#")
                    Node (tag: kParenGroup) {
                      (#'(' @20-21: "(")
                      Node (tag: kActualParameterPositionalList) {
                        Node (tag: kExpression) {
                          Node (tag: kReferenceCallBase) {
                            Node (tag: kReference) {
                              Node (tag: kLocalRoot) {
                                Node (tag: kUnqualifiedId) {
                                  (#SymbolIdentifier @21-22: "x")
                                }
                              }
                            }
                          }
                        }
                      }
                      (#')' @22-23: ")")
                    }
                  }
                }
                (#"::" @23-25: "::")
                Node (tag: kUnqualifiedId) {
                  (#SymbolIdentifier @25-29: "func")
                }
              }
            }
            Node (tag: kParenGroup) {
              (#'(' @29-30: "(")
              (#')' @30-31: ")")
            }
          }
        }
        (#';' @31-32: ";")
      }
```

The `#`'s direct parent is `kActualParameterList`, which should be sufficient to distinguish from other contexts.

fangism, 2020-05-01:
Another good test case to add to formatter_test.cc:

```systemverilog
class foo extends bar #(y);
endclass
```

should become:

```systemverilog
class foo extends bar#(y);
endclass
```

because `bar#(y)` contains actual type parameters.

fangism, 2020-05-01:
Actually, I stand corrected:

Our guidance says that parameterized type reference "foo #(...)" shall NOT be space-separated when it appears a part of a `::`-qualified name.  So the following are good:

```systemverilog
class foo extends bar #(...);  // not qualified
endclass

task t;
  bar #(y) z;  // not qualified, a data declaration
  pkg::bar#(y) z;  // qualified, a data declaration
  foo#(x)::func(...);  // qualified
  pkg::foo#(x)::func(...); // qualified
endtas
```

Fortunately, this is still do-able, just need to examine further up the context, checking for `kQualifiedId`.

The formatter was designed to support context-sensitive cases like this.

tgorochowik, 2020-06-22:
@wtatarski is working on this. For some reason I am unable to assign it to him.

mithro, 2020-06-22:
@wtatarski should have a pending invite now.

fangism, 2020-06-22:
Thanks @wtatarski !

commit 25623775 (2020-07-07): PR #339: Remove space before `#(`
Fixes #196
Fixes #287
GitHub PR https://github.com/google/verible/pull/339

Copybara import of the project:

  - 96ea5c8e2beac142a56af3275dc548eb3f6f38b2 Added left_context to AnnotateWithContextTestCase by Wojciech Tatarski <wtatarski@antmicro.com>
  - f3ba98a48489384281d2ee9dce736d95821f77a8 Remove space before `#(` by Wojciech Tatarski <wtatarski@antmicro.com>

Closes #339

PiperOrigin-RevId: 319452571
closed 2020-07-07 by hzeller