chipsalliance/verible #291: [git users] Lint only changed lines

msfschaffner, 2020-05-07:
It would be convenient for PR CI checks to have a similar feature as #206 for the Verible style linter as well.

msfschaffner, 2020-05-07:
b/155932004

fangism, 2020-05-07:
Very do-able.

For a cumulative-effect `--lines` flag, copy/refactor:
https://github.com/google/verible/blob/8bfe7a7f51a238a842d9c8300453f4d2122a963f/verilog/tools/formatter/verilog_format.cc#L52

Plumb through line ranges as an `IntervalSet` into
https://github.com/google/verible/blob/8bfe7a7f51a238a842d9c8300453f4d2122a963f/verilog/analysis/verilog_linter.h#L106

Re-use as-is:
https://github.com/google/verible/blob/master/verilog/tools/formatter/diff-to-changed-lines.awk

Modify:
https://github.com/google/verible/blob/master/verilog/tools/formatter/git-verilog_format.sh
to invoke `verilog_lint`.  

And of course, test.  :-)

fangism, 2020-05-19:
Do #292 before this one, so that the same solution there can be applied to the linter script.

fangism, 2020-08-04:
This script has a more modernized example of how to extract changed lines per file for a few different revision control systems.

https://github.com/google/verible/blob/master/verilog/tools/formatter/verible-verilog-format-changed-lines-interactive.sh