chipsalliance/verible #292: [git users] Support for arbitrary diffbase in git-verilog_format.sh

msfschaffner, 2020-05-07:
This is an enhancement request for adding support for arbitrary diffbases to the git-verilog_format.sh script (and the linter script, once that has been added https://github.com/google/verible/issues/291).

msfschaffner, 2020-05-07:
b/155932665

msfschaffner, 2020-05-07:
Here is an example for a CI check in OpenTitan that is run on all commits from the fork point up to HEAD:
https://github.com/lowRISC/opentitan/blob/d677d78836ccaa80725c7f0e3d0fa832a2037f86/azure-pipelines.yml#L106-L122
https://github.com/lowRISC/opentitan/blob/d677d78836ccaa80725c7f0e3d0fa832a2037f86/util/lint_commits.py#L137

fangism, 2020-08-04:
This script has an example of choosing a relative base:

https://github.com/google/verible/blob/master/verilog/tools/formatter/verible-verilog-format-changed-lines-interactive.sh