chipsalliance/verible #297: verilog_format: single statement not indented if not in begin/end block

hzeller, 2020-05-10:
Consider input

`testfile.v`
```Verilog
module foo ();
   always @(posedge clk or posedge rst)
     if (rst) state <= STATE_RESET;
     else state <= STATE_GO;
endmodule
```

It is not using begin/end inside the always block (not arguing if this is a good idea or not, but seen in code in the wild).
With that, `verilog_format` does not see the indentation need, and removes the indentation from the if-statement.

```Verilog
$ verilog_format testfile.v
module foo ();
  always @(posedge clk or posedge rst)
  if (rst) state <= STATE_RESET;
  else state <= STATE_GO;
endmodule
```

fangism, 2020-05-10:
The lack of begin/end is something for the style-linter to detect and suggest-fix.
But the missing indentation problem is confirmed.

fangism, 2020-05-10:
Suggest-fix general project is #284

fangism, 2020-05-10:
b/156188324

hzeller, 2020-05-11:
Well, in my personal Verilog style I actually do _not_ want begin/end in single-statement situations, so a lint suggested fix would be not useful in this case.
So this is why I want the formatter to honor the intend.

fangism, 2020-05-11:
The formatter promises to never make lexical changes, only whitespace, so you're safe.  :)

commit ca9d9892 (2020-05-21): Properly indent nested controlled contructs, w/ single-statement bodies.

fixes #297

PiperOrigin-RevId: 312210286
closed 2020-05-21 by hzeller