chipsalliance/verible #303: parse-as-class-body doesn't work

pdonahue-ventana, 2020-05-19:
Using prebuilt verible-v0.0-383-g0b0011b binaries, it looks like `verilog_syntax: parse-as-class-body` doesn't work because it complains about "non-class functions":
```
$ cat bar.svh
// verilog_syntax: parse-as-class-body
function int method();
  return 1;
endfunction
$ verible-v0.0-383-g0b0011b/bin/verilog_lint bar.svh
bar.svh:2:14: Explicitly define static or automatic lifetime for non-class functions [Style: function-task-explicit-lifetime] [explicit-function-lifetime]
```

fangism, 2020-05-19:
Thanks for the report.  This is a known limitation at this time in the style linter, but the parser is working as intended.  When using the alternate parsing directives, it 'fakes' context text to enable re-use of the existing top-level parsing.  (For class bodies, imagine prolog="class c;\n", epilog = "endclass\n".).  But then after that parsing, the fake context is discarded, by zooming into the syntax tree subnode that corresponded to the original text.  That is the point at which the fake context is lost, and the analyzed syntax tree is rooted at the class-body, rather than the class-declaration.

The question is, if we want to preserve some of the fake context, which is relevant to some rules like this one, it may be surprising to the user to find some references to code that was not entered (e.g. the fake class name).  We just need to be cautious.

fangism, 2020-05-19:
b/145160035 (internal reference)

fangism, 2020-05-19:
If you run `verilog_syntax --printtree` on your example, you can see this:

```
Parse Tree: 
Node (tag: kClassItems) {
  Node (tag: kFunctionDeclaration) {
    Node (tag: kFunctionHeader) {
      (#"function" @39-47: "function")
      Node (tag: kDataType) {
        Node (tag: kDataTypePrimitive) {
          (#"int" @48-51: "int")
          Node (tag: kPackedDimensions) {
          }
        }
      }
      Node (tag: kUnqualifiedId) {
        (#SymbolIdentifier @52-58: "method")
      }
      Node (tag: kParenGroup) {
        (#'(' @58-59: "(")
        (#')' @59-60: ")")
      }
      (#';' @60-61: ";")
    }
    Node (tag: kBlockItemStatementList) {
      Node (tag: kJumpStatement) {
        (#"return" @64-70: "return")
        Node (tag: kExpression) {
          Node (tag: kNumber) {
            (#TK_DecNumber @71-72: "1")
          }
        }
        (#';' @72-73: ";")
      }
    }
    (#"endfunction" @74-85: "endfunction")
  }
}
```

The root node is `kClassItems`.

pdonahue-ventana, 2020-05-21:
Thanks.  I'm just kicking the tires on verible so I can live with this limitation for now.

fangism, 2020-05-21:
> Thanks. I'm just kicking the tires on verible so I can live with this limitation for now.

Sure.  Keep on kicking!

theswit, 2022-01-28:
Note : the unit test for `parse-as-class-body` passes with or without the pragma:
https://github.com/chipsalliance/verible/blob/0834c7f8bd/verilog/analysis/verilog_analyzer_test.cc#L403