chipsalliance/verible #310: Support for naming conventions

zegervdv, 2020-05-28:
Are there any plans to add linting rules for naming conventions?

For example module inputs should start with `i_` or should end with `_i`.
Preferably this would be configurable using some kind of template.

fangism, 2020-05-28:
I'm hearing similar requests coming from some teams.  I acknowledge not everyone will share the same convention, but fortunately, we've introduced rule configurability so it should be possible to pass regular expressions to the rules that check conventions.

@msfschaffer I believe you wanted this too, right?

To point to one example:
https://github.com/google/verible/blob/3968647730731ee8533b75097245efc23a9271a8/verilog/analysis/checkers/struct_union_name_style_rule.cc#L77

The currently hardcoded style checks could come from rule configurations.

Here's an example of a configurable rule:
https://github.com/google/verible/blob/3968647730731ee8533b75097245efc23a9271a8/verilog/analysis/checkers/explicit_parameter_storage_type_rule.cc#L104

pawelsag, 2021-03-31:
Hi, so if I understand it correctly we would like to have the whole name to be configured with regex or just only some parts of the gathered name should be verified with regex?

What regex should I use, the one from the cpp standard?

zegervdv, 2021-03-31:
I think the whole name would allow for most flexibility. In the example I gave we want to check that input ports of the module always start with `i_`, but other organizations use the rule that inputs should end on `_i`. 
So, if possible, a full name regex match would allow for both, and many other rules you could think of. 

The configuration for that could then be something like: `\bi_.*` to match the start of a word. 

I'm using PCRE style regexes here but, for me at least, as long as it supports the basics and is well documented, the exact syntax is not that important. 

I guess that is up to @fangism (or other maintainers) to decide.

pawelsag, 2021-04-01:
Ok, thanks for the clarification.