chipsalliance/verible #312: New lint rule for checking port name postfixes

msfschaffner, 2020-05-29:
Some style guides such as the one used in OpenTitan require module ports to be postfixed with `*_i`, `*_o`, `*_io` depending on whether they are inputs/outputs/inouts.

https://github.com/lowRISC/opentitan/blob/f55d8ea669ff2bfa7e3ac16fab0674fb1d2ada4a/hw/ip/aes/rtl/aes.sv#L13-L26

We currently rely on manual code reviews for checking this, hence it would be useful if we could automate this by adding a lint rule that can check for such patterns. Note that it might be useful to make the postfix patterns parameterizable to accommodate different styles.

For an example of a module following this naming convention, see:
https://github.com/lowRISC/opentitan/blob/dbd19d6a82b004c7989eb5a3612c5bd020b4faa1/hw/ip/aes/rtl/aes_control.sv#L11-L100

msfschaffner, 2020-05-29:
b/157749206

fangism, 2020-05-29:
Agreed that the desired convention should be configurable.
Whether or not it needs to support fully general regexes is up for debate.

msfschaffner, 2020-05-29:
Full regex may be a bit of an overkill for our use case. But if you can leverage some synergies with other rules where you plan to add regex matching, it could make this rule a bit more generic.

fangism, 2020-09-19:
This can start with those hardcoded patterns.

PR #580 (2021-04-01): Add port name suffix lint rule
Fixes #312
closed 2021-04-01 by tgorochowik