chipsalliance/verible #313: align module named port connections

fangism, 2020-05-30:
Like this:

```systemverilog
module m;
  foo #(
    ...
  ) bar (
    .a   (a),
    .aa  (aa),
    .aaa (aaa)
  );
endmodule
```

If a port is abbreviated with just `.aa`, then there is no need for that line to align with anything else.

Recommend following the example:

https://github.com/google/verible/blob/85f8ff8134957d220400e29b05036318523ea22e/verilog/formatting/align.cc#L80

I expect the handling around new column markers in the case of named port connections to be simpler.

fangism, 2020-05-30:
b/152805837

Ciprian167, 2020-06-01:
I would like to give this a try. @fangism, could you please assign it to me?

fangism, 2020-06-01:
Assigned, thanks for volunteering.

fangism, 2020-06-05:
Quickly checking in, have you been able to make some progress or have any questions?

Ciprian167, 2020-06-09:
I didn't got the chance to start it yet. I will come back with questions soon.

fangism, 2020-06-09:
No problem.  Make sure you sync up through at least https://github.com/google/verible/commit/bdbc19b016898410e8eb3c0b327d822f1acbd994 because I've made some interface improvements that should help with controlling alignment.

fangism, 2020-06-19:
Just checking in again.  Have any questions?

Ciprian167, 2020-06-29:
> Just checking in again.  Have any questions?
Sorry I didn't had much time this past month. I'm giving it try today.

Ciprian167, 2020-07-14:
@fangism I've got a couple of questions:
1. Right now I implemented the named port scanner inside /verible/verilog/formatting/align.cc. Should I move it out in a separate file?

2. I'm trying to workout the following corner cases when interlacing wildcard connections, implicit connections and named connections. 
First, the following:
 ```systemverilog
module m;
    foo bar(.a(a), .aa, .bbb(b));
endmodule
```
Gets formatted:
```systemverilog
module m;
  foo bar (
      .a  (a),
      .aa,
      .bbb(b)
  );
endmodule
```
I guess `.a(a)` should not be aligned with `.bbb(b)`, since they are divided by an implicit connection, which is not aligned with any of them. I assume, I have to do something similar to `GetSubpartitionsBetweenBlankLines` and break partitions separated by blank lines, `.*`, and `.a`, right?

Second, the following:
 ```systemverilog
module m;
    foo bar(.a(a), .*, .bbb(b));
endmodule
```
Ouputs:
```systemverilog
module m;
  foo bar (
      .a  (a), .*,
      .bbb(b)
  );
endmodule
```
This feels like something doesn't work with `kFitOnLineElseExpand` policy. Even when not applying my alignment I get `.*` on the same line with .a(a).

fangism, 2020-07-14:
1. It's ok to keep Verilog-specific alignment code in verilog/formatting/align.cc, it was expected to grow.

2. (various)

For now, you may treat the `.aa` implicit named connection like a common 'row': it does not participate in alignment, but rows above and below it still align with each other.  Currently, a blank line is what divides alignment groups in module port declarations, and that's a reasonable start.  What you show actually looks acceptable, with `(a)` lining up with `(b)`.

As for `.*`, there is probably logic missing for making that occupy its own partition, which explains what you're seeing.

I examine `verible-verilog-format --show_token_partition_tree` on your example and see:

```
Full token partition tree:
{ ([<auto>], policy: always-expand) @{}
  { ([<auto>], policy: fit-else-expand) @{0}, (origin: "module m;
 ...;
endmodule")
    { ([module m ;], policy: fit-else-expand, (origin: "module m;")) }
    { (>>[<auto>], policy: fit-else-expand) @{0,1}, (origin: "foo bar(.a(..., .bbb(b));")
      { (>>[foo bar (], policy: fit-else-expand, (origin: "foo")) }
      { (>>>>>>[<auto>], policy: always-expand) @{0,1,1}, (origin: ".a(a), .*, .bbb(b)")
        { (>>>>>>[. a ( a ) , .* ,], policy: fit-else-expand, (origin: ".a(a)")) }     // OOPS: .* should have started its own partition
        { (>>>>>>[. bbb ( b )], policy: fit-else-expand, (origin: ".bbb(b)")) }
      }
      { (>>[) ;], policy: uninitialized) }
    }
    { ([endmodule], policy: always-expand, (origin: "endmodule")) }
  }
}
```

I would treat `.*` handling as a separate issue to be fixed in verilog/formatting/tree_unwrapper.cc.

For now, it is acceptable in this PR to leave `.*` not handled properly.  We can fix one thing at a time.

Thanks for continuing this!

fangism, 2020-07-14:
Separated the `.*` partitioning issue as #363

Ciprian167, 2020-07-15:
There is a similar problem when using  comments:
```systemverilog
module m;
    foo bar(
    .a(a),
    //.aa(aa), 
    .aaa(aaa));
endmodule
```
Outputs:
```systemverilog
module m;
  foo bar (
      .a  (a),
      //.aa(aa), .aaa(aaa)
  );
endmodule
```
Which can be problematic.

fangism, 2020-07-15:
That needs to be fixed by adding a case here:

https://github.com/google/verible/blob/1886f72085356d3169d35a8b496f97fe1c662959/verilog/formatting/tree_unwrapper.cc#L513

Let me try to fix this quickly today.

(Incidentally, the formatter runs a lexical equivalence check at the end and would bail out to prevent this sort of data corruption.)

Ciprian167, 2020-07-15:
I suppose we need a case for `NodeEnum::kPortActualList`. I can add it and commit the changes for this issue as well.

fangism, 2020-07-15:
Exactly.  I'll split this off as a separate issue, and assign it to you.  (I like keeping issues separate because it make it easier for developers to understand small directed fixes.)

fangism, 2020-07-15:
Hmm, I tried to run `verible-verilog-format --format_module_instantiations` on your example with comments, and am unable to reproduce the problematic output.  There may be some interaction with your ongoing work.  I can't file a separate issue without a standalone test case, but I'll let you continue to work on it.

Ciprian167, 2020-07-15:
You're right. I can see this only after changing this policy to `kTabularAlignment`
https://github.com/google/verible/blob/1886f72085356d3169d35a8b496f97fe1c662959/verilog/formatting/tree_unwrapper.cc#L919

commit 7f398147 (2020-07-22): Enable named ports alignment of instances by default.

fixes #313
issues #368

PiperOrigin-RevId: 322253604
closed 2020-07-22 by hzeller