chipsalliance/verible #318: align declared parameters inside #(...)

fangism, 2020-06-06:
Task:

Align parameter declarations only inside `#(...)` like so:

keyword | type [dimensions] | id | = default
------------ | -------------|-----|------
`parameter` | `int` | `foo` | `= 1;`
`localparam` | `int` | `foo` | `= 11111;`
` ` | `int` | `foo` | `= 1;`
`parameter` | `type` | `MyType` | `= int;`
`parameter` | `int [3:0]` | `foo_bars` | `= {...};`

See existing port declaration aligner in
https://github.com/google/verible/blob/master/verilog/formatting/align.cc

fangism, 2020-06-06:
b/158030314

Ciprian167, 2020-07-26:
I have a partial implementation for this. I'm trying to figure out a few cases mostly when using type declaration with packed dimensions. 

1. Should the formatter align packed dimensions like so?
```systemverilog
module foo #(
    type T  = int [  3:0], 
    type TT = bit [250:0]
);
endmodule
```

Implementing the packed dimensions alignment will make it difficult to align things like:
```systemverilog
module foo #( type T = int [3:0], type TT  = bit [250:0],  type TTT = C#(logic [128:0]));
endmodule
```
Which current implementation will format to :
```systemverilog
module foo #(
    type T   = int               [  3:0],
    type TT  = bit               [250:0],
    type TTT = C#(logic [128:0])
);
endmodule
```
This happens because of the way columns are created. Here is an excerpt from debug log.
```
end of FillAlignmentRow, row: < type | T | = int | [ | 3 : 0 | ] >
end of FillAlignmentRow, row: < type | TT | = bit | [ | 250 : 0 | ] >
end of FillAlignmentRow, row: < type | TTT | = C # ( logic [ 128 : 0 ] ) |  |  |  >
```

2. Another case I'm trying to figure out is, when using macros like so:
```systemverilog
module foo #(parameter int a = 2, parameter int aaa = `MACRO);
endmodule
```
The formatter outputs:
```systemverilog
module foo #(
    parameter int a   = 2,
    parameter int aaa = `MACRO
);
endmodule
```
Which I think it's decent, but I get the following error/warning:
```
Formatted output is lexically different from the input.    Please file a bug.  Details:
Mismatched token enums.  got: (MacroIdentifier) (#295: "`MACRO") vs. (MacroIdItem) (#297: "`MACRO")
First mismatched token [14]: (MacroIdentifier) (#295: "`MACRO") vs. (MacroIdItem) (#297: "`MACRO")
; problematic formatter output is
module foo #(
    parameter int a   = 2,
    parameter int aaa = `MACRO
);
endmodule
<<EOF>>
```

fangism, 2020-07-26:
> I have a partial implementation for this. I'm trying to figure out a few cases mostly when using type declaration with packed dimensions.
> 
> 1. Should the formatter align packed dimensions like so?
> 
> ```systemverilog
> module foo #(
>     type T  = int [  3:0], 
>     type TT = bit [250:0]
> );
> endmodule
> ```

This is acceptable.

> Implementing the packed dimensions alignment will make it difficult to align things like:
>
> ```systemverilog
> module foo #( type T = int [3:0], type TT  = bit [250:0],  type TTT = C#(logic [128:0])); [...]

This actually looks ok, it follows from the grammatic structure.  
`C#(logic[128:0])` is treated as a base type without dimensions.

Again, if you want to leave everything to the right of `=` as one column, that is fine too.

> 1. Another case I'm trying to figure out is, when using macros like so:
> 
> ```systemverilog
> module foo #(parameter int a = 2, parameter int aaa = `MACRO);
> endmodule
> ```
> 
> The formatter outputs:
> 
> ```systemverilog
> module foo #(
>     parameter int a   = 2,
>     parameter int aaa = `MACRO
> );
> endmodule
> ```
> 
> Which I think it's decent, but I get the following error/warning:
> 
> ```
> Formatted output is lexically different from the input.    Please file a bug.  Details:
> Mismatched token enums.  got: (MacroIdentifier) (#295: "`MACRO") vs. (MacroIdItem) (#297: "`MACRO")
> First mismatched token [14]: (MacroIdentifier) (#295: "`MACRO") vs. (MacroIdItem) (#297: "`MACRO")
> ; problematic formatter output is
> module foo #(
>     parameter int a   = 2,
>     parameter int aaa = `MACRO
> );
> endmodule
> <<EOF>>
> ```

Hmm, this seems like a separate bug in the equivalence verification logic.  I need to make it more forgiving.  If it is reproducible without your local changes, can you file this issue separately?

fangism, 2020-08-18:
Just checking in, have you been able to make any progress?  (avoiding the macro value example for a workaround)

fangism, 2020-08-20:
@Ciprian167 Sounds like you had work-in-progress a while ago.  Any updates or need help?

Ciprian167, 2020-08-21:
Ohh. I forgot I started this. Sorry. Let me brush up.

commit cf2df3df (2020-08-27): PR #412: Issue 318 parameters declaration alignment
First implementation for #318

GitHub PR https://github.com/google/verible/pull/412

Copybara import of the project:

  - 0754b93eda68ecea093e0b615860638deb315781 Merge remote-tracking branch 'google/master' into issue_3... by Ciprian <ciprian.antoci@gmail.com>
  - 5b2e2fa8fd388b4d510365ccfd3f1c92e1f0c1a8 First implementation for parameter declaration alignment. by Ciprian <ciprian.antoci@gmail.com>
  - f9be7453f0b9f00be438afef9aae7bc8e2ea2af5 Merge remote-tracking branch 'google/master' into issue_3... by Ciprian <ciprian.antoci@gmail.com>
  [... cut]
closed 2020-08-27 by hzeller