chipsalliance/verible #328: One line statements with comments

zegervdv, 2020-06-16:
I'm running some formatting tests with code from the lowRISC styleguide and I'm getting unexpected results.

Example code: 
```systemverilog
always_comb begin
unique case (state)
   STIDLE:              
      next_state = STA;  // case items at a block boundary without using begin
   STA:                 // and end.  Case items should fit on a single line, or
      next_state = STB;  // else the procedural block must have begin and end.
   STB: begin
      foo = bar;
      next_state = STIDLE;
   end
   default: begin
      next_state = STIDLE;
   end
endcase
end
```

After formatting I expect the single line expression after the first two cases to be moved to the same line.
This is the case for the first one, but the second one is left untouched (because of the comment after that line).
Also note the missing indentation on the line after `STA:`
 
Output:
```systemverilog
always_comb begin
  unique case (state)
    STIDLE: next_state = STA;  // case items at a block boundary without using begin
    STA:  // and end.  Case items should fit on a single line, or
    next_state = STB;  // else the procedural block must have begin and end.
    STB: begin
      foo = bar;
      next_state = STIDLE;
    end
    default: begin
      next_state = STIDLE;
    end
  endcase
end
```
I understand the comment after `STA` makes the correct action ambiguous: if the lines are merged, where should the first comment go?
Maybe the formatter could report an error on this, so the user can move the comment to a more appropriate location?

fangism, 2020-06-16:
The //comment between `STA:` and `next_state = STB;` prevents them from being moved onto the same line, the formatter does not re-order comment and non-comment tokens.

However, if you wrote:

```systemverilog
    STIDLE: next_state = STA;  // case items at a block boundary without using begin
    // and end.  Case items should fit on a single line, or
    STA:
    next_state = STB;  // else the procedural block must have begin and end.
```

that would be considered movable by the formatter.

Also note that at this time, the formatter doesn't support "re-flowing" of comments yet, that is moving around text among consective `//comment` lines.  It is a long-term goal, but today, the comments are treated atomically.  If reflowing is desired, I propose to file a separate request for it.

Observation: The line
`next_state = STB;` looks wrapped incorrectly, and (I think I know why).  We could focus this issue about the wrapping correctness.

zegervdv, 2020-06-16:
Understood, the re-flowing issue does not sound like a trivial problem :)

Do you think a linting rule to disallow comments before a single line statement would be acceptable to prevent (unintended) code like this? We want to avoid this code construct, but the comment line blocks the formatter from fixing it.

I agree the wrapping error is the main focus for this issue here.

fangism, 2020-06-16:
I have plans on how to handle re-flowing of comments, string-literals, multi-line macro definitions, but it also requires similar enhancements to the equivalence checker (verifies that the result of formatting is equivalent to the original).  It would require more changes to the detail-formatter because what was once atomic is now "soft".

I don't think discouraging a comment in that location is a good idea, it could be a good location to describe what is about to happen in the code.

```systemverilog
    STA:  // this describes what the following statements do, or what the label means
      next_state = ...;
```

"Attachment" of comments has some subtle heuristics.  I can summarize as: _"Comments that follow any code on the same line are considered attached to the code on the left.  Comments that occupy their own line will be formatted as if they were a syntactic item indented to the same level as the surrounding code."_ 

You should expect formatting output like:

```systemverilog
    STA:
      // this comment occupies a whole line and will remain as such after formatting, with proper indentation
      next_state = ...;
```

So, let's focus this bug on missing indentation of the `next_state = STB;` statement.