chipsalliance/verible #329: Check for style-compliant format-string numeric specifiers

fangism, 2020-06-17:
According to our style guide, we should always display numbers in the following manner:

```systemverilog
// good examples
$display("0x%0x", some_hex_value);
$display("0b%0b", some_binary_value);
$display("'h%0h", some_hex_value);
$display("'b%0b", some_binary_value);  // also acceptable
$display("%0d", some_decimal_value);  // no prefix wanted
```

```systemverilog
// bad examples
$display("%0x", some_hex_value);
$display("%0b", some_binary_value);
$display("0d%0d", some_decimal_value);
$display("'d%0d", some_decimal_value);
```

See:
https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#number-formatting

This includes any task function that is `$display`-like, including `$sformat`.

fangism, 2020-06-17:
b/120784977

fangism, 2020-06-17:
Possible approach might start with analyzing all string-literals (not just those in `$sformat` args):

```
if type(token) == TK_StringLiteral:
  for all occurrences of "%0h":
    if occurrence is not immediately preceded by "'h":
      report violation
  repeat for other radix markers...
```

ayaankhan98, 2020-07-02:
@fangism i would like to give it a try, if you can provide some help

fangism, 2020-07-02:
> @fangism i would like to give it a try, if you can provide some help

Always happy to answer questions.

Some tips to get started:

To examine a token stream: `verible-verilog-syntax --printtokens FILE`

Here is one example of a lint rule that only examines lexical tokens:

https://github.com/google/verible/blob/master/verilog/analysis/checkers/macro_name_style_rule.h

fangism, 2020-07-10:
Just checking in.  Any questions I can answer to help?

commit 172dbadf (2020-10-14): PR #524: Add numeric format string style checker
Fixes #329

GitHub PR https://github.com/google/verible/pull/524

Copybara import of the project:

  - 38d16a0b2c6a25e7c6a332d40ba76ca8e12ab7ba Add numeric format string style checker by Lukasz Dalek <ldalek@antmicro.com>
  - 5368b124a927c5c547859099b1e153e43dd9391d Merge 38d16a0b2c6a25e7c6a332d40ba76ca8e12ab7ba into e6486... by Lukasz Dalek <48219411+ldalek-antmicro@users.noreply.github.com>

Closes #524

PiperOrigin-RevId: 336918731
closed 2020-10-14 by hzeller