chipsalliance/verible #330: Inconsistent whitespace in [SIZE-1:0] vector declarations

pdonahue-ventana, 2020-06-17:
The whitespace surrounding operators in vector declarations is inconsistent between vectors of built-in types and vectors of user-defined types:
```
$ verible/bin/verible-verilog-format --version
v0.0-447-gb8504ca
Commit	2020-06-17 11:01:41 -0700
Built	2020-06-17T18:54:03Z
$ verible/bin/verible-verilog-format foo.sv
typedef logic boolean_t;
parameter int SIZE = 3;
module foo;
  logic [SIZE-1:0] a;
  boolean_t[SIZE - 1:0] b;
endmodule
```

I would expect both of them to either be `[SIZE-1:0]` or `[SIZE - 1:0]` (preferably the former) rather than a mixture.  The space before the opening bracket is also inconsistent.

Thanks.

fangism, 2020-06-17:
Confirmed.  At this time, our style guidance makes an exception to allow compact or padded binary expressions inside `[]` for declaration dimensions, and I've consciously chosen to leave these all of these spaces alone for the moment to reduce churn.

Here is the excerpt of code that controls that spacing behavior:

https://github.com/google/verible/blob/b8504ca0596871b6791db5695dd93b39fa44c6aa/verilog/formatting/token_annotator.cc#L152

The missing space after `boolean_t` looks like a problem though, I agree there should be a space after it.

fangism, 2020-06-17:
Quick triage:

`verible-verilog-format --show_token_partition_tree --show_inter_token_info`:

```
Full token partition tree:
{ ([<auto>], policy: always-expand) @{}
  { ([<_0,0>typedef <_1,9>logic <_1,9>boolean_t <_0,19>;], policy: fit-else-expand, (origin: "typedef logic boolean_t;")) }
  { ([<_1,2>parameter <_1,9>int <_1,11>SIZE <_1,14>= <_1,11>3 <_0,19>;], policy: fit-else-expand, (origin: "parameter int SIZE = 3;")) }
  { ([<auto>], policy: fit-else-expand) @{2}, (origin: "module foo;...  endmodule")
    { ([<_1,2>module <_1,11>foo <_0,21>;], policy: fit-else-expand, (origin: "module foo;")) }
    { (>>[<auto>], policy: always-expand) @{2,1}, (origin: "logic [SIZE...E - 1:0] b;")
      { (>>[<_1,4>logic <_1,24>[ <_0,25>SIZE <pre>- <pre>1 <_0,25>: <_0,25>0 <_0,25>] <_1,15>a <_0,23>;], policy: fit-else-expand, (origin: "logic [SIZE-1:0] a;")) }
      { (>>[<_1,6>boolean_t <_0,39>[ <_0,21>SIZE <_1,33>- <_1,25>1 <_0,21>: <_0,21>0 <_0,21>] <_1,15>b <_0,23>;], policy: fit-else-expand, (origin: "boolean_t[SIZE - 1:0] b;")) }      // after boolean_t, should be <_1,###> not <_0,###>
    }
    { ([<\n>endmodule], policy: always-expand, (origin: "endmodule")) }
  }
}
```

We just need to make the min-spacing value between `ID` and `[` to be 1 in a declarative context.  The formatter supports context-aware policies.

fangism, 2020-06-17:
b/159256218 (internal reference)

fangism, 2020-06-17:
A suggested test case for whomever takes this:

Expect formatter to produce:

```systemverilog
module n;
  type_t [w[1]:w[0]] foo;
endmodule
```

Note that spacing comes after `type_t`, but not `w`.