chipsalliance/verible #335: Missing indentation on named arguments of functions

zegervdv, 2020-06-19:
When formatting function calls with 'named arguments' (not sure what they are called) the formatter does not indent the arguments in the same way as for unnamed arguments.

Example:
```systemverilog
  always_comb begin
    value = function_name(
    .long_parameter(8'hA),
    .another_parameter(signal_123),
    .parameter_three(foobar)
    );
  end
```

For unnamed arguments (if the line were long enough) this would be:
```systemverilog
  always_comb begin
    value = function_name(8'hA, signal_123,
                                         foobar);
  end
```

Some indentation before each argument would make it easier readable.

fangism, 2020-06-19:
Confirmed.  Just haven't handled this case yet in

https://github.com/google/verible/blob/master/verilog/formatting/tree_unwrapper.cc

closed 2021-05-18 by hzeller

wtatarski, 2021-05-19:
I'm testing my fix with some parts of code from OpenTitan. I see that `new`  constructor:
```
class dv_base_mem extends uvm_mem;
  function void configure();
    begin
      mam = new(
      get_full_name,
      cfg,
      this);
    end

    if (hdl_path != "")
      add_hdl_path_slice(hdl_path, -1,
                         -1);
  endfunction : configure

endclass
```
taken from https://github.com/lowRISC/opentitan/blob/master/hw/dv/sv/dv_base_reg/dv_base_mem.sv
column_limit is set to 40

wtatarski, 2021-05-27:
After applying #837 fix formatting looks better for above case:
```
class dv_base_mem extends uvm_mem;
  function void configure();
    begin
      mam =
          new(get_full_name, cfg, this);
    end

    if (hdl_path != "")
      add_hdl_path_slice(hdl_path, -1,
                         -1);
  endfunction : configure

endclass
```

wtatarski, 2021-05-27:
Before:
```
class aes_common_vseq extends aes_base_vseq;
  virtual function void ctrl_reg_map_invalid_value(ref bit [TL_DW-1:0] val);
    ctrl_shadowed_val = get_csr_val_with_updated_field(
        ral.ctrl_shadowed.operation, ctrl_shadowed_val, operation_val
    );
    ctrl_shadowed_val = get_csr_val_with_updated_field(
        ral.ctrl_shadowed.mode, ctrl_shadowed_val, mode_val
    );
    ctrl_shadowed_val = get_csr_val_with_updated_field(
        ral.ctrl_shadowed.key_len, ctrl_shadowed_val, key_len_val
    );
    ctrl_shadowed_val = get_csr_val_with_updated_field(
        ral.ctrl_shadowed.manual_operation, ctrl_shadowed_val, manual_operation_val
    );

    if (!$cast(mode_e, mode_val)) begin
      ctrl_shadowed_val = get_csr_val_with_updated_field(
          ral.ctrl_shadowed.mode, ctrl_shadowed_val, AES_NONE
      );
    end
    if (!$cast(key_len_e, key_len_val)) begin
      ctrl_shadowed_val = get_csr_val_with_updated_field(
          ral.ctrl_shadowed.key_len, ctrl_shadowed_val, AES_256
      );
    end
  endfunction
endclass
```
After
```
class aes_common_vseq extends aes_base_vseq;
  virtual function void ctrl_reg_map_invalid_value(ref bit [TL_DW-1:0] val);
    ctrl_shadowed_val = get_csr_val_with_updated_field(ral.ctrl_shadowed.operation,
                                                       ctrl_shadowed_val, operation_val);
    ctrl_shadowed_val =
        get_csr_val_with_updated_field(ral.ctrl_shadowed.mode, ctrl_shadowed_val, mode_val);
    ctrl_shadowed_val =
        get_csr_val_with_updated_field(ral.ctrl_shadowed.key_len, ctrl_shadowed_val, key_len_val);
    ctrl_shadowed_val = get_csr_val_with_updated_field(ral.ctrl_shadowed.manual_operation,
                                                       ctrl_shadowed_val, manual_operation_val);

    if (!$cast(mode_e, mode_val)) begin
      ctrl_shadowed_val =
          get_csr_val_with_updated_field(ral.ctrl_shadowed.mode, ctrl_shadowed_val, AES_NONE);
    end
    if (!$cast(key_len_e, key_len_val)) begin
      ctrl_shadowed_val =
          get_csr_val_with_updated_field(ral.ctrl_shadowed.key_len, ctrl_shadowed_val, AES_256);
    end
  endfunction
endclass
```

tgorochowik, 2021-05-27:
@imphil could you please confirm that the formatting in the post above looks ok? (in terms of function argument alignment)

PR #802 (2021-05-18): Fix indentation on named arguments of functions
Fixes #335