chipsalliance/verible #336: Comments after enum values are formatted in new line

hzeller, 2020-06-21:
(not sure if this is a duplicate of some existing issue)

Consider something like this

```verilog
  typedef enum {
    CMD_NO_OP =  0,      // Get fifo free slots
    CMD_STATUS = 1,      // get status word
    CMD_WRITE_FIFO = 2   // send segments to fifo
  } command_t;
```

All these lines fit comfortably in the allotted line length, yet the formatter changes this to

```verilog
  typedef enum {
    CMD_NO_OP = 0
    ,  // Get fifo free slots
    CMD_STATUS = 1
    ,  // get status word
    CMD_WRITE_FIFO = 2  // send segments to fifo
  } command_t;
```

It looks like it is eager to break up around the comma, as the last line is left as-is.

hzeller, 2020-06-23:
```
$ verible-verilog-format --show_token_partition_tree /tmp/enum.sv
Full token partition tree:
{ ([<auto>], policy: always-expand) @{}
  { ([<auto>], policy: fit-else-expand) @{0}, (origin: "typedef enu... command_t;")
    { ([typedef enum {], policy: fit-else-expand) }
    { (>>[<auto>], policy: always-expand) @{0,1}, (origin: "CMD_NO_OP =...TE_FIFO = 2")
      { (>>[CMD_NO_OP = 0], policy: fit-else-expand, (origin: "CMD_NO_OP =  0")) }
      { (>>[, // Get fifo free slots], policy: uninitialized) }
      { (>>[CMD_STATUS = 1], policy: fit-else-expand, (origin: "CMD_STATUS = 1")) }
      { (>>[, // get status word], policy: uninitialized) }
      { (>>[CMD_WRITE_FIFO = 2 // send segments to fifo], policy: fit-else-expand, (origin: "CMD_WRITE_FIFO = 2")) }
    }
    { ([} command_t ;], policy: fit-else-expand) }
  }
}
 typedef enum {
    CMD_NO_OP =  0,      // Get fifo free slots
    CMD_STATUS = 1,      // get status word
    CMD_WRITE_FIFO = 2   // send segments to fifo
  } command_t;
```

fangism, 2020-06-23:
The `, //...` starting its own token partition node is wrong.

hzeller, 2020-08-05:
One-stop self-contained partition tree debug invocation, directly from bazel:

```shell
bazel run //verilog/tools/formatter:verible-verilog-format -- --show_token_partition_tree - <<EOF
  typedef enum {
    CMD_NO_OP =  0,      // Get fifo free slots
    CMD_STATUS = 1,      // get status word
    CMD_WRITE_FIFO = 2   // send segments to fifo
  } command_t;
EOF
```

fangism, 2020-08-14:
b/164297034

hzeller, 2020-08-14:
Also useful: switching on vlogging for the relevant code; here with a narrowed-down example that shows that only code with comma followed by a comment is resulting in an uninitialized policy, while an enum assignment just followed by comma is properly assigned.

```
TF_CPP_VMODULE="tree_unwrapper=3" bazel run //verilog/tools/formatter:verible-verilog-format -- --show_token_partition_tree - <<EOF
  typedef enum {
    CMD_NO_OP =  0,      // Get fifo free slots
    CMD_FOO =  42,
    CMD_WRITE_FIFO = 2   // send segments to fifo
  } x_t;
EOF
```

```
Output:
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kDescriptionList
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kDescriptionList
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kTypeDeclaration
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kTypeDeclaration
I common/formatting/tree_unwrapper.cc:309] VisitIndentedChildren, new child node {0,0}: [], policy: fit-else-expand
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#"typedef" @2-9: "typedef")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#"typedef" @2-9: "typedef")
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kDataTypePrimitive
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kDataTypePrimitive
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kEnumType
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kEnumType
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#"enum" @10-14: "enum")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#"enum" @10-14: "enum")
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kBraceGroup
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kBraceGroup
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#'{' @15-16: "{")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#'{' @15-16: "{")
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kEnumNameList
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kEnumNameList
I common/formatting/tree_unwrapper.cc:309] VisitIndentedChildren, new child node {0,1,0}: >>[], policy: fit-else-expand

I verilog/formatting/tree_unwrapper.cc:571] Visit node: kEnumName
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kEnumName
I common/formatting/tree_unwrapper.cc:309] VisitIndentedChildren, new child node {0,1,0,0}: >>[], policy: fit-else-expand
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#SymbolIdentifier @21-30: "CMD_NO_OP")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#SymbolIdentifier @21-30: "CMD_NO_OP")
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kTrailingAssign
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kTrailingAssign
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#'=' @31-32: "=")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#'=' @31-32: "=")
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kExpression
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kExpression
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kNumber
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kNumber
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#TK_DecNumber @34-35: "0")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#TK_DecNumber @34-35: "0")
I verilog/formatting/tree_unwrapper.cc:1283] ReshapeTokenPartitions node: kEnumName
I verilog/formatting/tree_unwrapper.cc:1713] end of ReshapeTokenPartitions node: kEnumName
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#',' @35-36: ",")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#',' @35-36: ",")

I verilog/formatting/tree_unwrapper.cc:571] Visit node: kEnumName
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kEnumName
I common/formatting/tree_unwrapper.cc:309] VisitIndentedChildren, new child node {0,1,2,0}: >>[], policy: fit-else-expand
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#SymbolIdentifier @69-76: "CMD_FOO")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#SymbolIdentifier @69-76: "CMD_FOO")
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kTrailingAssign
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kTrailingAssign
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#'=' @77-78: "=")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#'=' @77-78: "=")
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kExpression
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kExpression
I verilog/formatting/tree_unwrapper.cc:571] Visit node: kNumber
I verilog/formatting/tree_unwrapper.cc:592] SetIndentationsAndCreatePartitions node: kNumber
I verilog/formatting/tree_unwrapper.cc:1718] Visit leaf: (#TK_DecNumber @80-82: "42")
I verilog/formatting/tree_unwrapper.cc:1810] end of Visit leaf: (#TK_DecNumber @80-82: "42")
[... 36 lines cut ...]
I verilog/formatting/tree_unwrapper.cc:1713] end of ReshapeTokenPartitions node: kDataTypePrimitive
[... cut ...]
    { (>>[<auto>], policy: always-expand) @{0,1}, (origin: "CMD_NO_OP =...TE_FIFO = 2")
      { (>>[CMD_NO_OP = 0], policy: fit-else-expand, (origin: "CMD_NO_OP =  0")) }
      { (>>[, // Get fifo free slots], policy: uninitialized) }
      { (>>[CMD_FOO = 42 ,], policy: fit-else-expand, (origin: "CMD_FOO =  42")) }
      { (>>[CMD_WRITE_FIFO = 2 // send segments to fifo], policy: fit-else-expand, (origin: "CMD_WRITE_FIFO = 2")) }
    }
    { ([} x_t ;], policy: fit-else-expand) }
  }
}
```
Anyway, this is an example that highlights that the comma is only

fangism, 2020-08-14:
At this point, the options for fixing are (without digging in):

1) prevent the comma-partition from being created
1) reshape/adjust/merge/fixup after the fact

commit 6bd02180 (2020-08-18): PR #406: Include comma as a valid trailing delimiter for enum lists.
Fixes #336

GitHub PR https://github.com/google/verible/pull/406

Copybara import of the project:

  - 0d56276a4ce29b0ec907c0f5b8f9616a9ce5662d Include comma as a valid trailing delimiter for enum lists. by Henner Zeller <h.zeller@acm.org>
  - 121cfbe1275f8fea5bd8ff3e1afb6a847b5c786e Address review comments. Add tests for tree unwrapper and... by Henner Zeller <h.zeller@acm.org>
  - 45c6ed6ace56146f7951dbaff5a50bbc0f049744 Add enum value inside ifdef region. by Henner Zeller <h.zeller@acm.org>

Closes #406

PiperOrigin-RevId: [... cut]
closed 2020-08-18 by hzeller