chipsalliance/verible #340: deploy to edaplayground.com or our own playground?

fangism, 2020-06-23:
To raise-awareness and lower the barrier to testing Verible, should we push to deploy at edaplayground.com or consider our own playground?  @mithro WDYT?

When we were brainstorming project ideas, here are some of the ideas we had for tools/panels in the interface:

* pure syntax check
* syntax tree dump `verible-verilog-syntax --printtree`, but with navigable/expandable tree view
* style linter
* formatter demo

svenka3, 2020-11-08:
I would be thrilled to see an online Verible on its own site (without other commercial simulators, that complicates some end-users' use cases - I personally have heard many users saying that). On what else to add to such interface - how about a document generator (especially for SV/UVM) - similar to Doxygen/NaturalDocs. And a way to export/download formatted code (Consider user has code that's unformatted, uses this site to format, then pull the formatted code back to his/her use).

Thanks
Srini

mithro, 2020-11-08:
On the topic of documentation (which is mostly unrelated to verible), we are currently using [Sphinx](https://www.sphinx-doc.org/en/master/) for documenting our Verilog projects (which is very popular in the Python community). The [sphinx-verilog-domain](https://github.com/SymbiFlow/sphinx-verilog-domain) module adds Verilog / SystemVerilog support and it goes well with the [sphinxcontrib-hdl-diagrams](https://github.com/SymbiFlow/sphinxcontrib-hdl-diagrams) module for generating cool RTL diagrams automagically, [Symbolator](https://github.com/SymbiFlow/symbolator) for generating module diagrams and [sphinxcontrib-wavedrom](https://pypi.org/project/sphinxcontrib-wavedrom/) for creating nice waveform diagrams.

fangism, 2020-11-08:
Nevertheless, comment-based documentation extraction is a very reasonable feature set to ask for, do file a separate issue for that.  It could be further enhanced by our ongoing search index extraction project using Kythe as well.