chipsalliance/verible #341: Support for comments in waiver files

imphil, 2020-07-02:
Waiver files for Verible lint (passed with `--waiver_files`) do not (seem to) support comments. Can that support be added, in line with the comment support in rules files, which was added in https://github.com/google/verible/issues/274?

Comments in waiver are essential to give a bit of motivation why a waiver is necessary (and to add a license header).

fangism, 2020-07-02:
What style of comments would you like?  `#`?  `//`?

Would need to start by adding a comment token pattern here:

https://github.com/google/verible/blob/master/common/analysis/command_file.lex

and make sure it gets filtered out before reaching the parser.

imphil, 2020-07-02:
I don't mind; `#` would be consistent with the rule config files (see  #274 / https://github.com/google/verible/commit/7d55b5b320b8d60d0a1a8924ea9f8fa02a50551d), so I'd opt for that.

commit b1027eed (2020-08-10): PR #376: Added support for comments in waiver files
Fixes #341

GitHub PR https://github.com/google/verible/pull/376

Copybara import of the project:

  - 153dbb3d6308fadf3964056a6225ce39762f5f5f Added support for comments in waiver files by Wojciech Tatarski <wtatarski@antmicro.com>

Closes #376

PiperOrigin-RevId: 325879021
closed 2020-08-10 by hzeller