chipsalliance/verible #343: Support lint waivers without regex and line

imphil, 2020-07-02:
I have the following Verible lint error that I want to waive:

```
../../../rtl/ibex_register_file_ff.sv:13:8: Declared module does not match the first dot-delimited component of file name: "ibex_register_file_ff" [Style: file-names] [module-filename]
```

This waiver doesn't need a regex or line since it applies to the whole file. However, a waiver like this fails:

```
waive --rule=module-filename --location="ibex_register_file_ff\.sv"
```

```
E common/analysis/lint_waiver.cc:450] ../../../lint/verible_waiver.vbw:1:68: command error: Insufficient waiver configuration
```

Would it be possible to make both `--line` and `--regex` optional?

--- 

As a workaround, I'm currently using

```
waive --rule=module-filename --regex=".*" --location="ibex_register_file_ff\.sv"
```

The equivalent Verilator waiver looks like this: https://github.com/lowRISC/ibex/blob/master/lint/verilator_waiver.vlt#L28

fangism, 2020-07-02:
@tgorochowik Would this be easy (and make sense) to support?

tgorochowik, 2020-07-03:
Yes, it would absolutely make sense to add this.

Initially we did not support `--location` in the waiver files so we could do this by disabling a rule globally with regular linter config.
With the `--location` flag we can disable the rule for a specific file only so it makes sense to add this.

I also think it should be very easy to add.
The first step would be to allow `can_use_regex`  and `can_use_lineno` to be both false here:
https://github.com/google/verible/blob/4d6929e0e059c7735b5023095a94120a3d833dd6/common/analysis/lint_waiver.cc#L352-L379

And then add some code to handle this at the end of this snippet (just before returning `okStatus`).
Since the waiver is based on line numbers the simplest way would be to just waive it for all lines (`TextStructure` is one of the arguments in this function so it should be easy to do).

commit 617b4041 (2020-07-29): PR #362: Added support for lint waivers without regex and line
Fixes #343

GitHub PR https://github.com/google/verible/pull/362

Copybara import of the project:

  - 808f2b22aac7304482ea360dea3e021ee76c58c5 Added support for lint waivers without regex and line by Wojciech Tatarski <wtatarski@antmicro.com>

Closes #362

PiperOrigin-RevId: 323870869
closed 2020-07-29 by hzeller