chipsalliance/verible #344: Verible lint: Allow `5'b?` in rule undersized-binary-literal

imphil, 2020-07-02:
The undersized-binary-literal lint rule allows undersized zero constants (e.g. `5'b0`), but doesn't allow undersized "don't care" constants (e.g. `5'b?`). 

```
../../../rtl/ibex_tracer_pkg.sv:300:68: Binary literal 5'b? is shorter than its declared width: 5. [Style: number-literals] [undersized-binary-literal]
```

The workaround in code is rather easy (do `5'h?`), but it seems inconsistent to not allow `?` to be undersized, but `0`.

fangism, 2020-07-02:
Is there any semantic difference between: `5'b?` and `5'b?????` ?  [language lawyer]

What should happen if the literal is a mix of `01?` ?

imphil, 2020-07-02:
> Is there any semantic difference between: 5'b? and 5'b????? ? [language lawyer]

They are identical from my reading of the spec. 

SV2017 says (pages 73 and 74):

"If the size of the unsigned number is smaller than the size specified for the literal constant, the unsigned number shall be padded to the left with zeros. If the leftmost bit in the unsigned number is an x or a z, then an x or a z shall be used to pad to the left, respectively. (...) When used in a number, the question mark (?) character is a SystemVerilog alternative for the z character."

Just as another data point: Verilator also has a similar warning, and accepts `5'b?`.

> What should happen if the literal is a mix of 01? ?

Well, the synthesis would pad with 0s. But for the style linter, I'd expect the undersized-binary-literal error.

commit 61e17ba4 (2021-01-26): PR #647: Lint: Allow `5'b?` in rule undersized-binary-literal
Closes #344

GitHub PR https://github.com/google/verible/pull/647

Copybara import of the project:

  - 5ea68104572ef1c979d8bf4377ac55061ec866e2 Lint: Allow `5'b?` in rule undersized-binary-literal by Mariusz Glebocki <mglebocki@antmicro.com>
  - 400b2a6e2b6df445459c0f5c495e305713553b0f Merge 5ea68104572ef1c979d8bf4377ac55061ec866e2 into d4fa1... by Mariusz Glebocki <mglb@arccos-1.net>

Closes #647

PiperOrigin-RevId: 353915048
closed 2021-01-26 by hzeller