chipsalliance/verible #345: Travis build fails on Ubuntu 20.04 LTS (Focal Fossa) due to missing python-gflags package

mithro, 2020-07-03:
From https://github.com/google/verible/pull/342#issuecomment-653420870_
> `apt-get install python-gflags`  # fails on Ubuntu 20.04 LTS (Focal Fossa) google/python-gflags#44
> `apt-get install -y     bazel     build-essential     gcc     g++     file     lsb-release     python-gflags     wget`
> [__E: Unable to locate package python-gflags__](https://travis-ci.org/github/google/verible/jobs/704574359#L1304)

mithro, 2020-07-03:
We can probably solve this by instead installing python-gflags using pip.

fangism, 2020-07-03:
According to the front-page:
https://github.com/google/python-gflags

this should now be part of absl: https://github.com/abseil/abseil-py

fangism, 2020-07-09:
Fixed by https://github.com/google/verible/commit/a1f54754ba26815bbfddf0280e9cb165f2b9d579 ?