chipsalliance/verible #346: Verible format: Some token partitions failed to complete within the search limit

imphil, 2020-07-03:
STR:
- Get `prim_lfsr.sv` from the lowrisc/ibex repository: https://github.com/lowRISC/ibex/blob/master/vendor/lowrisc_ip/prim/rtl/prim_lfsr.sv.
- Run verible format on it:
   ```
   $ verible-verilog-format vendor/lowrisc_ip/prim/rtl/prim_lfsr.sv
   ```

That's the output I get (removed the source which is output as well):

```
vendor/lowrisc_ip/prim/rtl/prim_lfsr.sv: *** Some token partitions failed to complete within the search limit:
>>[localparam logic [ 63 : 0 ] GAL_XOR_COEFFS [ 61 ] = '{ 64 'h 9 , 64 'h 12 , 64 'h 21 , 64 'h 41 , 64 'h 8E , 64 'h 108 , 64 'h 204 , 64 'h 402 , 64 'h 829 , 64 'h 100D , 64 'h 2015 , 64 'h 4001 , 64 'h 8016 , 64 'h 10004 , 64 'h 20013 , 64 'h 40013 , 64 'h 80004 , 64 'h 100002 , 64 'h 200001 , 64 'h 400010 , 64 'h 80000D , 64 'h 1000004 , 64 'h 2000023 , 64 'h 4000013 , 64 'h 8000004 , 64 'h 10000002 , 64 'h 20000029 , 64 'h 40000004 , 64 'h 80000057 , 64 'h 100000029 , 64 'h 200000073 , 64 'h 400000002 , 64 'h 80000003B , 64 'h 100000001F , 64 'h 2000000031 , 64 'h 4000000008 , 64 'h 800000001C , 64 'h 10000000004 , 64 'h 2000000001F , 64 'h 4000000002C , 64 'h 80000000032 , 64 'h 10000000000D , 64 'h 200000000097 , 64 'h 400000000010 , 64 'h 80000000005B , 64 'h 1000000000038 , 64 'h 200000000000E , 64 'h 4000000000025 , 64 'h 8000000000004 , 64 'h 10000000000023 , 64 'h 2000000000003E , 64 'h 40000000000023 , 64 'h 8000000000004A , 64 'h 100000000000016 , 64 'h 200000000000031 , 64 'h 40000000000003D , 64 'h 800000000000001 , 64 'h 1000000000000013 , 64 'h 2000000000000034 , 64 'h 4000000000000001 , 64 'h 800000000000000D } ;], policy: fit-else-expand, (origin: "localparam ...00000000D};")
>>[localparam logic [ 167 : 0 ] FIB_XNOR_COEFFS [ 166 ] = '{ 168 'h 6 , 168 'h C , 168 'h 14 , 168 'h 30 , 168 'h 60 , 168 'h B8 , 168 'h 110 , 168 'h 240 , 168 'h 500 , 168 'h 829 , 168 'h 100D , 168 'h 2015 , 168 'h 6000 , 168 'h D008 , 168 'h 12000 , 168 'h 20400 , 168 'h 40023 , 168 'h 90000 , 168 'h 140000 , 168 'h 300000 , 168 'h 420000 , 168 'h E10000 , 168 'h 1200000 , 168 'h 2000023 , 168 'h 4000013 , 168 'h 9000000 , 168 'h 14000000 , 168 'h 20000029 , 168 'h 48000000 , 168 'h 80200003 , 168 'h 100080000 , 168 'h 204000003 , 168 'h 500000000 , 168 'h 801000000 , 168 'h 100000001F , 168 'h 2000000031 , 168 'h 4400000000 , 168 'h A000140000 , 168 'h 12000000000 , 168 'h 300000C0000 , 168 'h 63000000000 , 168 'h C0000030000 , 168 'h 1B0000000000 , 168 'h 300003000000 , 168 'h 420000000000 , 168 'h C00000180000 , 168 'h 1008000000000 , 168 'h 3000000C00000 , 168 'h 6000C00000000 , 168 'h 9000000000000 , 168 'h 18003000000000 , 168 'h 30000000030000 , 168 'h 40000040000000 , 168 'h C0000600000000 , 168 'h 102000000000000 , 168 'h 200004000000000 , 168 'h 600003000000000 , 168 'h C00000000000000 , 168 'h 1800300000000000 , 168 'h 3000000000000030 , 168 'h 6000000000000000 , 168 'h D800000000000000 , 168 'h 10000400000000000 , 168 'h 30180000000000000 , 168 'h 60300000000000000 , 168 'h 80400000000000000 , 168 'h 140000028000000000 , 168 'h 300060000000000000 , 168 'h 410000000000000000 , 168 'h 820000000001040000 , 168 'h 1000000800000000000 , 168 'h 3000600000000000000 , 168 'h 6018000000000000000 , 168 'h C000000018000000000 , 168 'h 18000000600000000000 , 168 'h 30000600000000000000 , 168 'h 40200000000000000000 , 168 'h C0000000060000000000 , 168 'h 110000000000000000000 , 168 'h 240000000480000000000 , 168 'h 600000000003000000000 , 168 'h 800400000000000000000 , 168 'h 1800000300000000000000 , 168 'h 3003000000000000000000 , 168 'h 4002000000000000000000 , 168 'h C000000000000000018000 , 168 'h 10000000004000000000000 , 168 'h 30000C00000000000000000 , 168 'h 600000000000000000000C0 , 168 'h C00C0000000000000000000 , 168 'h 140000000000000000000000 , 168 'h 200001000000000000000000 , 168 'h 400800000000000000000000 , 168 'h A00000000001400000000000 , 168 'h 1040000000000000000000000 , 168 'h 2004000000000000000000000 , 168 'h 5000000000028000000000000 , 168 'h 8000000004000000000000000 , 168 'h 18600000000000000000000000 , 168 'h 30000000000000000C00000000 , 168 'h 40200000000000000000000000 , 168 'h C0300000000000000000000000 , 168 'h 100010000000000000000000000 , 168 'h 200040000000000000000000000 , 168 'h 5000000000000000A0000000000 , 168 'h 800000010000000000000000000 , 168 'h 1860000000000000000000000000 , 168 'h 3003000000000000000000000000 , 168 'h 4010000000000000000000000000 , 168 'h A000000000140000000000000000 , 168 'h 10080000000000000000000000000 , 168 'h 30000000000000000000180000000 , 168 'h 60018000000000000000000000000 , 168 'h C0000000000000000300000000000 , 168 'h 140005000000000000000000000000 , 168 'h 200000001000000000000000000000 , 168 'h 404000000000000000000000000000 , 168 'h 810000000000000000000000000102 , 168 'h 1000040000000000000000000000000 , 168 'h 3000000000000006000000000000000 , 168 'h 5000000000000000000000000000000 , 168 'h 8000000004000000000000000000000 , 168 'h 18000000000000000000000000030000 , 168 'h 30000000030000000000000000000000 , 168 'h 60000000000000000000000000000000 , 168 'h A0000014000000000000000000000000 , 168 'h 108000000000000000000000000000000 , 168 'h 240000000000000000000000000000000 , 168 'h 600000000000C00000000000000000000 , 168 'h 800000040000000000000000000000000 , 168 'h 1800000000000300000000000000000000 , 168 'h 2000000000000010000000000000000000 , 168 'h 4008000000000000000000000000000000 , 168 'h C000000000000000000000000000000600 , 168 'h 10000080000000000000000000000000000 , 168 'h 30600000000000000000000000000000000 , 168 'h 4A400000000000000000000000000000000 , 168 'h 80000004000000000000000000000000000 , 168 'h 180000003000000000000000000000000000 , 168 'h 200001000000000000000000000000000000 , 168 'h 600006000000000000000000000000000000 , 168 'h C00000000000000006000000000000000000 , 168 'h 1000000000000100000000000000000000000 , 168 'h 3000000000000006000000000000000000000 , 168 'h 6000000003000000000000000000000000000 , 168 'h 8000001000000000000000000000000000000 , 168 'h 1800000000000000000000000000C000000000 , 168 'h 20000000000001000000000000000000000000 , 168 'h 48000000000000000000000000000000000000 , 168 'h C0000000000000006000000000000000000000 , 168 'h 180000000000000000000000000000000000000 , 168 'h 280000000000000000000000000000005000000 , 168 'h 60000000C000000000000000000000000000000 , 168 'h C00000000000000000000000000018000000000 , 168 'h 1800000600000000000000000000000000000000 , 168 'h 3000000C00000000000000000000000000000000 , 168 'h 4000000080000000000000000000000000000000 , 168 'h C000300000000000000000000000000000000000 , 168 'h 10000400000000000000000000000000000000000 , 168 'h 30000000000000000000006000000000000000000 , 168 'h 600000000000000C0000000000000000000000000 , 168 'h C0060000000000000000000000000000000000000 , 168 'h 180000006000000000000000000000000000000000 , 168 'h 3000000000C0000000000000000000000000000000 , 168 'h 410000000000000000000000000000000000000000 , 168 'h A00140000000000000000000000000000000000000 } ;], policy: fit-else-expand, (origin: "localparam
...000000000};")
*** end of partially formatted partition list
[other error status]
```

I get a similar error with https://github.com/lowRISC/ibex/blob/master/vendor/lowrisc_ip/prim/rtl/prim_cipher_pkg.sv

```
$ verible-verilog-format --version
v0.0-459-g4d6929e
Commit  2020-06-30
Built   2020-07-02T09:54:39Z
```

imphil, 2020-07-03:
Another file with the same symptom: https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv

dxie-tt, 2024-12-18:
I found a work around by setting `--column_limit` with a large value

zapta, 2025-07-09:
Here is another file that fails

```
module pll (clkout, lock, clkin);

output clkout;
output lock;
input clkin;

wire clkoutp_o;
wire clkoutd_o;
wire clkoutd3_o;
wire gw_gnd;

assign gw_gnd = 1'b0;

rPLL rpll_inst (
    .CLKOUT(clkout),
    .LOCK(lock),
    .CLKOUTP(clkoutp_o),
    .CLKOUTD(clkoutd_o),
    .CLKOUTD3(clkoutd3_o),
    .RESET(gw_gnd),
    .RESET_P(gw_gnd),
    .CLKIN(clkin),
    .CLKFB(gw_gnd),
    .FBDSEL({gw_gnd,gw_gnd,gw_gnd,gw_gnd,gw_gnd,gw_gnd}),
    .IDSEL({gw_gnd,gw_gnd,gw_gnd,gw_gnd,gw_gnd,gw_gnd}),
    .ODSEL({gw_gnd,gw_gnd,gw_gnd,gw_gnd,gw_gnd,gw_gnd}),
    .PSDA({gw_gnd,gw_gnd,gw_gnd,gw_gnd}),
    .DUTYDA({gw_gnd,gw_gnd,gw_gnd,gw_gnd}),
    .FDLY({gw_gnd,gw_gnd,gw_gnd,gw_gnd})
);

defparam rpll_inst.FCLKIN = "27";
defparam rpll_inst.DYN_IDIV_SEL = "false";
defparam rpll_inst.IDIV_SEL = 8;
defparam rpll_inst.DYN_FBDIV_SEL = "false";
defparam rpll_inst.FBDIV_SEL = 19;
defparam rpll_inst.DYN_ODIV_SEL = "false";
defparam rpll_inst.ODIV_SEL = 16;
defparam rpll_inst.PSDA_SEL = "0000";
defparam rpll_inst.DYN_DA_EN = "false";
defparam rpll_inst.DUTYDA_SEL = "1000";
defparam rpll_inst.CLKOUT_FT_DIR = 1'b1;
defparam rpll_inst.CLKOUTP_FT_DIR = 1'b1;
defparam rpll_inst.CLKOUT_DLY_STEP = 0;
defparam rpll_inst.CLKOUTP_DLY_STEP = 0;
defparam rpll_inst.CLKFB_SEL = "internal";
defparam rpll_inst.CLKOUT_BYPASS = "false";
defparam rpll_inst.CLKOUTP_BYPASS = "false";
defparam rpll_inst.CLKOUTD_BYPASS = "false";
defparam rpll_inst.DYN_SDIV_SEL = 2;
defparam rpll_inst.CLKOUTD_SRC = "CLKOUT";
defparam rpll_inst.CLKOUTD3_SRC = "CLKOUT";
defparam rpll_inst.DEVICE = "GW2AR-18C";

endmodule 
```