chipsalliance/verible #347: Verible formatter: Formatted output is lexically different from the input (Mismatched token enums.)

imphil, 2020-07-03:
STR:
- Get https://github.com/lowRISC/ibex/blob/master/rtl/ibex_alu.sv
- Run 
   ```
   verible-verilog-format rtl/ibex_alu.sv
   ```

Output (stripped down to error message)

```
rtl/ibex_alu.sv: Formatted output is lexically different from the input.    Please file a bug.  Details:
Mismatched token enums.  got: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#745: ")")
First mismatched token [4059]: (')') (#41: ")") vs. (MacroCallCloseToEndLine) (#745: ")")
```

```
$ verible-verilog-format --version
v0.0-459-g4d6929e
Commit  2020-06-30
Built   2020-07-02T09:54:39Z
```

hankhsu1996, 2022-07-06:
Got the same error. Reproduce with the following code:

```systemverilog
`define LOG2(x) $clog2(x)
module foo;
  logic [`LOG2(4):0] bar;
endmodule
```

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

Current `ibex_alu.sv` from lowRISC formats with exit 0 and converges.